gijzelaerr/python-snap7 #42: Different behavior of read_area on windows and linux. read_area throws ValueError

oldfellow, 2015-03-31:
Hi,
Environment:
OS (Linux and Windows 7 ) : 64bit.
snap7-full-1.3.0.tar.gz
python-snap7-0.4.tar.gz

I have different behavior executing the same query on windows and linux.
I want to read 6 Objects from DB2 
read_area(132, 2, 0, 6)

The result on Windows is fine:
00 AB CD 11 B9 01    # This matches to the data in the plc.

On Linux I get the error:
File "/usr/local/lib/python2.6/site-packages/snap7/client.py", line 210, in read_area
    return bytearray(data)
ValueError: byte must be in range(0, 256)

For debugging I add the following line in front of the return in line 210:
print "".join("%02x " % b for b in data)

This gives me the Information that on linux the bytestream looks like this:
00 -55 -33 11 -47 01
And "-55" of course throws the ValueError.

When I run the same query with snap7 ( cpp or plain-c ), I get the correct values on windows and linux. Therefore I guess the issue must be somewhere in the python interface.

Any idea how to solve this issue?

Because I'm running python 2.6 I need to modify the util.py, because in python 2.6 the Collections doesn't provide OrderedDict.
I have installed ordereddict with pip and modified line 11 in util.py:
# from Collections import OrderedDict
import ordereddict
Maybe this creates the issue. But I don't know how to solve, because I can't change to python 2.7.

Thanks in advance.

gijzelaerr, 2015-03-31:
Hi OldFellow,
Thanks for reporting this issue.

The second issue is a separate issue, can you open a new issue for that?

gijzelaerr, 2015-03-31:
It looks like a signed vs unsigned issue to me @spreeker , what do you think? It is quite weird though, I think `read_area` is one of the most used functions in python-snap7, so it should have shown up before?

spreeker, 2015-03-31:
try to work with the "db_read" function! does it also happen there? 

I will add to the util.py file.
```
try:
    # try with the standard library
    from collections import OrderedDict
except ImportError:
    # fallback to Python 2.6-2.4 back-port
    from ordereddict import OrderedDict
```

gijzelaerr, 2015-03-31:
Please keep the python2.6 discussion here https://github.com/gijzelaerr/python-snap7/issues/44

spreeker, 2015-03-31:
@oldfellow please teast the "db_read" and an older snap7 to rule out all options. snap7 1.3 came with some endian awareness..

Snap7 1.3.0 is out !!!
(Platform and comfort release)
 
Snap7 now is runtime endian-aware.
(Runtime = No compilers switches are needed to work with Big-Endian CPUs)

gijzelaerr, 2015-03-31:
i've updated the package in the launchpad PPA (Ubuntu) and homebrew (OS X)

oldfellow, 2015-04-01:
db_read has the same behavior like read_area.
Which snap7 should I give a try?

spreeker, 2015-04-01:
Any lower than 1.3 I find the whole issue really weird.

What kind of values are you trying to read?  Are only some values wrong and
others good?

gijzelaerr, 2015-04-01:
did the signature of the function change to unsigned or something since 1.3.0? I can't check this now, in conference

oldfellow, 2015-04-01:
With snap7-1.2.1 I have exactly the same behavior.

I guess this must be related to the python wrapper. Using the same snap7.so ( from Version 1.3.0 or 1.2.1 ) with cpp or plain-c works fine.

oldfellow, 2015-04-01:
this is the the content of the plc:
DB2:
Byte 0 - 3 Real: 12.34
Byte 4 - 5 WORD: 0xABCD
Byte 6 - 7 INT: 4537
Byte 8 - 9 WORD: 0xFF00

( I use this funtion to print the raw data in client.py line 109 )
print "".join("0x%02x " % b for b in data)

In Windows Environment the result of db_read and read_area is equal:
0x41 0x45 0x70 0xa4 0xab 0xcd 0x11 0xb9 0xff 0x00
On Linux the result look like this:
0x41 0x45 0x70 0x-5c 0x-55 0x-33 0x11 0x-47 0x-1 0x00

As you can see in the last WORD, the 0xff is "translated" into 0x-1.
I agree, that this seems to be a signed / unsigned error. But at what point?

oldfellow, 2015-04-01:
When I add the following lines to client.py I can resolve this issue:
1. To client.py I add: 
   import binascii
2. Before the return of the function db_read, I add the following:
     hex_data = binascii.hexlify( data )
     data = binascii.unhexlify( hex_data )
     return bytearray(data)

I don't really understand, why converting two times solves this issue.
Any idea?

gijzelaerr, 2015-04-01:
it could be a bug in snap7, where the type of the byte is not defined as signed and the compiler functions differently on the two platforms.

gijzelaerr, 2015-04-01:
ah right, it works correctly if you write a C program

spreeker, 2015-04-01:
@oldfellow  Could you find out if and what the difference is of the data1 before and the data2 after the 
hexilfy steps?

hex_data = binascii.hexlify( data1 )
data2 = binascii.unhexlify( hex_data )
return bytearray(data)

spreeker, 2015-04-01:
Could you try to replace the python2.6 ctypes extension with a newer version?

oldfellow, 2015-04-15:
Sorry, because I'm working in a blackbox environment I can't change any python evironment.
How can I find out a difference bevore and after the hexlify? What PyCode should I use?

gijzelaerr, 2016-12-30:
hi @oldfellow this is quite an old issue now, sorry about that. Are you still experiencing these problems? Did you manage in the end?

closed 2017-02-08 by gijzelaerr