gijzelaerr/python-snap7 #44: Add Python2.6 support?

gijzelaerr, 2015-03-31:
@oldfellow reported:
"""
Because I'm running python 2.6 I need to modify the util.py, because in python 2.6 the Collections doesn't provide OrderedDict.
I have installed ordereddict with pip and modified line 11 in util.py:
```
from Collections import OrderedDict
import ordereddict
```
Maybe this creates the issue. But I don't know how to solve, because I can't change to python 2.7.

"""
If this is the only python2.6 compatibility then maybe it is a good idea to add Python2.6 support. But probably not, since it makes it harder to have a mixed python2/python3 code base.

spreeker, 2015-03-31:
I added the 
```
try:                                                                                                                                          |      def test_read_multi_vars(self):
    # try with the standard library                                                                                                           |          db = 1
    from collections import OrderedDict                                                                                                       |  
except ImportError:                                                                                                                           |          # build and write test values
    # fallback to Python 2.6-2.4 back-port                                                                                                    |          test_value_1 = 129.5
    from ordereddict import OrderedDict 
```

to util.py and that should be enough.

gijzelaerr, 2015-03-31:
so we support python2.6 now? are the no other problems?

gijzelaerr, 2015-08-29:
lets just close this issue, python2.6 support is nice but shouldn't be a priority.

closed 2015-08-29 by gijzelaerr