gijzelaerr/python-snap7 #45: Snap 7 server: python

joerig, 2015-03-31:
How to use the server function? There isn't any documentation on how to do this.
I've created a server object plc and I'm trying to create a register area for DB50.

spreeker, 2015-03-31:
Well you can look at the Test code for examples.

There should be enough information.
And you might want to add the documentation to the project.  It is after
all an open source project maintained in our free time

Cheers

Stephan.

gijzelaerr, 2015-03-31:
Hi Joerig,

The server module is mostly intended and uses for testing purposes, so we didn't really document it much.  But the Python API is quite similar to the C API which is described in the snap7 documentation PDF. If some things are missing you can find it there. and indeed, if you have some spare time feel free to do a pull request to add the relevant sections to the server API :)

closed 2015-03-31 by gijzelaerr