gijzelaerr/python-snap7 #46: S7-200/Logo PLCs Can't Connect

SimplyAutomationized, 2015-04-11:
Cannot connect to my S7-200 PLC using the following code:

import snap7
plc = snap7.client.Client()
plc.set_connection_params("10.0.0.250",10,10)
plc.connect("10.0.0.250",0,1)

In the Client class the connect function uses the library.ConnectTo(...). Which is not going to work for logos or the 200 series.  There needs to be a library.Cli_Connect().  

I was able to connect to the S7-200 using C# the following code :
//S7-200 example
            Snap7.S7Client plc = new Snap7.S7Client();
            plc.SetConnectionParams("10.10.55.250", 0x1100, 0x1100);
            Snap7.S7Client.S7CpuInfo info = new S7Client.S7CpuInfo();
            plc.Connect();
            Console.Write("Connected:{0}",plc.Connected());
            Console.ReadLine();
            plc.Disconnect();
            //S7-1200 example
            Snap7.S7Client plc1200 = new Snap7.S7Client();
            //no connection params
            plc1200.ConnectTo("10.0.0.251",0,1);
            Console.Write("Connected:{0}", plc1200.Connected());
            Console.ReadLine();
            plc1200.Disconnect();

I tried getting connected to the S7-200 using this:\

plc.library.Cli_SetConnectionParams(plc.pointer,c_char_p(six.b("10.0.0.250")),c_uint16(10),c_uint16(10))
plc.library.Cli_Connect()

But i get a segmentation fault. :-(. 
any help would be appreciated. If u need to borrow a test plc let me know.

spreeker, 2015-04-13:
Well I'm happy to help out but I don't have a S7-200. So if I could borrow one...
Also I looks to me that it should be not to hard to implement the missing functions looking at the other functions that are already there..

I'm happy to help out. Where in the world are you located?

gijzelaerr, 2015-08-29:
as I pointed out in your PR, as far as I know you can just set the parameters before using connect(), connect to behaves the same but has no defaults.

https://github.com/gijzelaerr/python-snap7/pull/47

closed 2015-08-29 by gijzelaerr