gijzelaerr/python-snap7 #48: Partner is not connecting

june9713, 2015-06-15:
Nice to see you.
And I'm very appriciate for your awsome Snap7 Projects.
I am getting very much of help from your documentation.
Now I sucessfully send and recieve datas form plc via your
rich examples.(partenr)
So, It's means My PLC setting is correct.
but I couldnt success partner data send and recieve with
my python code.

(python 3.4 64bit , windows 7 64bit , Visual studio 2013)
##################################################
from snap7 import partner
def h2d(num): # convert hex string to integer
return int(num,16)
ptn = partner.Partner()
ptn_stat = ptn.create(1)
print(ptn_stat)
ptn.start_to("192.29.31.253","192.29.31.254",h2d("0300"),h2d("1004"))
print ( ptn.get_status() ) 
###########################
I expacted returns "3" but, "1" is comming.
My Local IP is :192.29.31.253 , TSAP = 03.00
and My PLC IP is: 192.29.31.254 TSAP = 10.04
So I convert TSAP to inteager. for example
0x0300 is 768 , and 0x1004 is 4100.
and I was wating status 3 from PLCb but still 1.
I read your reference doccumentation but, I couldnt find
mention about python code.
So I ask to you courteously to let me know how send and recieve
data with python code please.
Any tiny help for me will be very huge help for me.
Im very poor in English. Please understand my poor expression.
Thanks.
Best Regards Jay.

spreeker, 2015-06-15:
Hi.  There is example source  code.  Also c heck out the the utility code written to  con very data from python data types and to plc data  types

spreeker, 2015-06-15:
Hi.  There is example source  code.  Also c heck out the the utility code written to  con very data from python data types and to plc data  types

https://github.com/gijzelaerr/python-snap7/blob/master/example/example.py

june9713, 2015-06-15:
thank you very much spreeker.
I aleady read that but, That is about client connection.
I tried to find Partner Example but I couldnt.
And also,
There is test_partner.py but, actually, I dont understand How test_partner.py works.
So I tried connect by reference manual and http://python-snap7.readthedocs.org/
Is "example.py" include partner section?
Please help me.

spreeker, 2015-06-15:
I never had the need to use the partner connection. So it is very well possible those functions are not implemented correctly. Or not implemented at all! I would go into the source code and check the client code and the partner code and maybe do it yourself and contribute back the code :)

june9713, 2015-06-15:
Thanks Spreeker.
I know, my effort is not much...
Reason why I need to use partner connection is,
I wanna use BSend(FB12) , BRecieve(FB13) to exchange data.
and, I aleady sucessfully exchage data by client connection.
client works very well. it means it implemented  very well.( I think)
Anyway....
I need to go back to the source code of patner.py and drill it.
That should be best way to find way.
Thanks a lot. If I find something, I will post.

gijzelaerr, 2015-06-15:
note that python-snap7 is quite a thin wrapper around the original snap7 wrapper. maybe the snap7 library doc will help!

june9713, 2015-06-15:
Thansk gijzelaerr.
I will read snap7 library and doucumentation. 
Thanks.

june9713, 2015-06-15:
partner.start_to( ) functions has 4 parameters.
3rd,4th is TSAP in Integer.
If TSAP is 03.00 , Integer value is 768.
Am I right?

wilkpio, 2015-06-16:
Indeed partner implementation is not really complete. I tried that few months ago and I faced the same problem as une9713. I'll provide you with pull request that implements partner in a moment...

gijzelaerr, 2015-06-16:
@wilkpio that would be awesome! we (core developers) never used the partner code, so never tested it.

It would be even more awesome if you could keep the quality of python-snap7 highish and add doc and some tests!

june9713, 2015-06-16:
@wilkpio Oh, i appriciate to you much!!
I'm waiting your documentation!
I will try analysis and help other userslike you. Thanks again!!

gijzelaerr, 2015-08-29:
ok this is not really an issue anymore now. The partner implementation is in a PR.

closed 2015-08-29 by gijzelaerr