gijzelaerr/python-snap7 #51: Reading from PLC | db_read ArgumentError

grawee, 2015-08-19:
Dear Snap7 developers and users,

I have a problem with the "db_read(db_number, start, size)" function. The exact problem is with the "start" variable, to which I have to give the start address
of the variable inside the plc which I wish to read. If I set an integer value of this var the program runs without issues, however I do have some variables
inside the PLC with offsets like 4.1, 8.4 etc. So they are not integer values. When I set these offsets as "start" the function doesn't work and stops with the following error:

[image]

Here is a part of my program I use to read from the PLC:

[image]

Is there any way to set non-rounded "Start" values?

I develop the program under ubuntu in python language.

Thank you for the snap7 project
Also thank you in advance for the helps and tips!

wilkpio, 2015-08-19:
@grawee python-snap7 is just a ctypes wrapper around snap7 c library. 
The python function `db_read()` just calls `Cli_DBRead()` function from c library. 
According to header file (snap7.h) the "start" augment is declared as integer. 
See: https://github.com/SCADACS/snap7/blob/master/examples/cpp/snap7.h#L436

The types must match. Therefore you have to use integer as a start parameter in `db_read()` and read whole byte(s) from PLC. 
You should be able to read individual bits from returned bytearray with `snap7.util.get_bool().` 

See example:
```python
reading = plc.db_read(31, 120, 1)  # read 1 byte from db 31 staring from byte 120
snap7.util.get_bool(reading, 0, 5)  # read a value of fifth bit
```

gijzelaerr, 2015-08-20:
Thanks @wilkpio for answering this question.

grawee, 2015-08-24:
Thanls for the answer @wilkpio. It's working fine. I have one more question or problem..
So far, I used the db_write(db_number, start, data) command to write to the PLC. The problem is the same with this function as well, when I try to set float offset value for start. So, I tried to use the set_bool(_bytearray, byte_index, bool_index, value), but I don't know what I need to set to "_bytearray". Can you show me an example for this function.

Maybe good to know, when I want to write in PLC, I read from MySQL database, so I get a bool from SQL what I want to read in PLC.

spreeker, 2015-08-24:
Dear @grawee  

# The numbers are not FLOATS they look like floats but they are definatly NOT floats.

The smallest thing you can read from a PLC is a BYTE with snap7. One byte is 8 bits. 8 boolean values   fit in a byte. on position 0 till position 7.  BYTEINDEX.BITINDEX. =! FLOAT.
00000000 = 1 BYTE 

1 byte can be used to store 8 boolean values. Packing so many values in a byte is done to so save precious PLC space and memory.

## So how to change a bool value?

1)  Read out the _bytearray your boolean values are in. 
2) In this byte_array you need to find the byte (byteindex) 
3) Find the bit in this byte (bitindex) so you can 
4) set the correct value with utility function set_bool which does all the hard work.
set_bool(_bytearray = data which you read out before.., byte_index, bool_index, value),
5) write back the changed bytearray back to the PLC.

A round trip wil take 5ms.

The example code 

https://github.com/gijzelaerr/python-snap7/blob/master/example/

should help out.

hope this helps :)

closed 2015-08-24 by spreeker

kesharinandan1993, 2019-10-17:
Is there any way to write integer/string in PLC db..... Any link or help would be appreciated....

Mutexman, 2021-11-25:
Good day!
Please tell me how to use snap7.util.get_bool () in C ++. Why can't I figure it out myself! And is there any documentation and where can I see it?

swamper123, 2021-11-25:
@Mutexman 
1) This isue is closed.
2) Before you start opening a new issue, pls. use [Discussions](https://github.com/gijzelaerr/python-snap7/discussions) for questions (so we can seperate it)
3) This repo is a python layer around snap7 ...I don't know what you mean with "how to use this in C++"???

KeshariGupta-ML, 2021-11-25:
I have done through python.. I didn't use c++ directly

Mutexman, 2021-11-25:
Is there a similar C ++ layer around snap7?

Mutexman, 2021-11-25:
I understand correctly, this was used
https://python-snap7.readthedocs.io/en/1.0/_modules/snap7/util.html#get_bool
?
It turns out that you can rewrite this code in C ++ and there will be a wrapper in C ++.