gijzelaerr/python-snap7 #54: set_string not working correctly

zappfinger, 2015-10-16:
I try to use set_string in a byte array, but only the first 2 characters are written correctly, the rest is garbled. For example, when I use: set_string(db10, 0, 'abcd')
and I read it back with get_string, I get: 'ab@I'
Also, the docstring mentions a size argument that is not present..

zappfinger, 2015-10-16:
Sorry, my fault, I was doing something wrong. Ignore this.

spreeker, 2015-10-21:
I spend significant time making sure writing strings is done safely. I had one bug once by not checking the length and if the string fits in the intended space and overwriting some extra memory in the plc. oops..

closed 2015-10-21 by spreeker

zappfinger, 2015-10-21:
Are you aware that there are two types of strings in the S7 series? The String type is preceded by 2 bytes. Byte one is the maximum string length, byte 2 is the actual string length. I believe your string function does nothing with byte one, only byte 2 is written?
But the other type is a character array, that can start anywhere in a DB and has no preceding bytes...
This type has no functions for it to read or write...

spreeker, 2015-10-22:
no I had no need for the character array so i have not encountered it.

The reason I don't write the first byte is is that the first byte value (max_size) is set by the siemens step7 project and snap7 / python should under no circumstance change the first byte!!
By setting the actual string size it was still needed for the functions to clear's out the rest of the bytes with ' ' empty space to make it work correctly and prevent weird string values to show 'newvalue_partoflargeroldstring' on the siemen screens.

The max size is given to the function set_string for two reasons. 1 to make sure you know about what you put in and 2 set_string function among others is user by the DB class to edit large a object. The DB object expects a specification / layout of the data and in it the max string size is also specified. 
The DB object makes it possible to use a DB in a PLC like a Dict in Python but it is rather hard to setup correctly and I don't know how to make it any more easy..