gijzelaerr/python-snap7 #56: can't find library on python3 windows 8.1

LavaLumbert, 2015-11-17:
Hi,

i am new on github, so please be gentle if i am at the wrong place for my question. 

i try to install snap7 (to read from a S7-1200) with it's python-snap7 0.4 wrapper but i get always a traceback with the following simple code.

    from time import sleep
    import snap7
    from snap7.util import *
    import struct

    plc = snap7.client.Client()

Traceback:
```
    Traceback (most recent call last):
    File "Y:\Lonnox\Projekte\Bibliothek\Python und SPS\S7-1200 Test.py", line 6, in <module>
    plc = snap7.client.Client()
    File "C:\Python34\lib\site-packages\snap7\client.py", line 30, in __init__
    self.library = load_library()
    File "C:\Python34\lib\site-packages\snap7\common.py", line 54, in load_library
    return Snap7Library(lib_location).cdll
    File "C:\Python34\lib\site-packages\snap7\common.py", line 46, in __init__
    raise Snap7Exception(msg)
    snap7.snap7exceptions.Snap7Exception: can't find snap7 library. If installed, try running ldconfig
```

idconfig is only available on Linux, right ?
The steps i do to install snap7 and python wrapper are:
* Download snap7 from sourceforge and copy snap7.dll and snap7.lib to system32 folder of windows 8.1
* Install wrapper by using pip install python-snap7 with no errors.
How to install snap7 on windows correctly ?

i am using and familar with python3 / 32bit and can edit the source to work with python3, but i need a starting point how to set and load the needed library.

gijzelaerr, 2015-11-17:
Windows can't find the DLL somehow. Are you sure you are using a 32bit version of Python?

Actually, if I remember correctly, on windows your system path is the same as your library path. So you just need to make sure if that the library is somewhere in your system path.

LavaLumbert, 2015-11-19:
Tanks for your answer and sorry for the late reply. I need some try and error experiments to find a solution. 
First Part of solution is to add the location of the copied snap7.dll and .lib files to windows enviroment variables. Alternatively you can copy the files to your Python install folder if you have checked the "add path" option while installing Python.
The Second part is to choose the 32bit files if you have a 32bit Pyhton. That is still the case if you use Win 8 64bit.
See the picture for details:
[image]

closed 2015-11-19 by LavaLumbert

gijzelaerr, 2015-11-19:
good you solved this. Yes, *or* you need a 64 bit python + 64 bit snap7 dll, *or* you need to keep everything 32. Indeed you can run 32 bit binaries on windows 64, but you need to make sure all libraries are 32 bit then.

BoodyV, 2020-03-09:
Hi, 
I have the same issue. i'm trying to follow these instructions, but when i download SNAP7 from sourceforge the file snap7.lib is not included.
How can i find this file?