gijzelaerr/python-snap7 #58: Read Input from Siemens LOGO! 8

Astrolor, 2016-04-26:
Hi Everybody

I've been trying to read ou the state of the digital inputs of my LOGO 8 module but it didn't work...
Has anybody an idea which value I have to put for the "area" ?

For example I want to read out the input byte for the inputs I0 to I7 (VM range 1024.0 - 1024.7):

>>> import snap7
>>> plc = snap7.client.Client()
>>> plc.set_connection_params('192.168.0.5',0x0300,0x0200)
>>> plc.connect('192.168.0.5',0,1)
>>> plc.get_connected()
True
>>> byte = plc.read_area(DB1.X1024, 0, 0, 1)
Traceback (most recent call last):
  File "<pyshell#5>", line 1, in <module>
    byte = plc.read_area(DB1.X1024, 0, 0, 1)
NameError: name 'DB1' is not defined
>>> 

I would appreciate any help! Thanks.
Best regards
Lorenz

spreeker, 2016-04-26:
Well read the source code. The plc has different area's.

https://github.com/gijzelaerr/python-snap7/blob/113b53495f357cab96d94d483a5f1815a6912596/snap7/snap7types.py

and test code:

https://github.com/gijzelaerr/python-snap7/blob/113b53495f357cab96d94d483a5f1815a6912596/test/test_client.py

        def test_read_area(self):
                area = snap7.snap7types.areas.DB
                dbnumber = 1
                amount = 1                   # number of bytes
                start = 1                        # starting point
                self.client.read_area(area, dbnumber, start, amount)

spreeker, 2016-04-26:
And more example code.

https://github.com/gijzelaerr/python-snap7/tree/master/example

closed 2016-04-26 by spreeker