gijzelaerr/python-snap7 #61: Missing the wrap of Cli_Connect()

davenardella, 2016-09-09:
To connect to a LOGO/S7200 must be used Cli_SetConnectionParams followed by Cli_Connect() (without parameters).

gijzelaerr, 2016-09-09:
Did you try `set_connection_params()` and `connect()`?

https://github.com/gijzelaerr/python-snap7/blob/master/snap7/client.py#L339

https://github.com/gijzelaerr/python-snap7/blob/master/snap7/client.py#L110

davenardella, 2016-09-09:
Hi, 
I don't know python, an user requested to me how to connect snap7 to a LOGO.
The connect that I see in client.py wraps Cli_ConnectTo() which has IP,Rack and Slot as parameters.

Sometime (LOGO/S7200) you need to build the TSAPs yourself in accord to the hardware configuration because there is no Rack/Slot concept and this is made with set_connection_params().

After this you need to call Cli_Connect() (without parameters) which uses the TSAPs built.

I expected to find something like this:

`@error_wrap
    def connect(self):
        """
        Connect to a S7 server using previous set parameters
        """
        logger.info("connecting")

        return self.library.Cli_Connect(
            self.pointer)`

Maybe I missed this in your code ? (I'm not skilled at all in python)

gijzelaerr, 2016-09-10:
I recommend you get yourself familiar with python and have a look at the examples:

https://github.com/gijzelaerr/python-snap7/blob/master/example/example.py

basically you just want to initialise the client class, call the `set_connection_params` client method to set the connection parameters and then call `.connect()` to open the connection.

gijzelaerr, 2016-12-30:
closing this issue, please reopen if you have further issues related to errors in python-snap7.

closed 2016-12-30 by gijzelaerr

Feinwerk, 2017-05-01:
Hi 

I have the same problem as the user ( davenardella) before the .connect() function is missing. 
It's not implemented in the code. I tried to get it to run but i am not skilled enought in Python.
I am using a Siemens logo which i have to connect with Tsap interface.

thanks

Alex