gijzelaerr/python-snap7 #62: Reading timers and counters

Lyckyrpi, 2016-10-26:
Hi there,
I got to write and read DB, MK, PE, PA using a siemens plc. It works with booleans, byte, words, reals and dwords.
Now I can't get to read a timer, using the same read_area function.
result=plc.read_area(areas["TM"],0,13,STWLTimer)
where 13 is the timer I want to read. In the plc program is S_EVERZ 13

It always says "CPU: address out of range".

Where am I wrong ?

gijzelaerr, 2016-10-26:
Sorry @Lyckyrpi i don't know. I don't think this is an issue with python-snap7, but you never know. I would suggest check if you are reading the correct area, start byte and size.

Lyckyrpi, 2016-10-26:
Yes the area are correct, I think there something wrong in the python wrapper. BTW, normally I don't need to read timers or counters.

gijzelaerr, 2016-10-26:
ok well i will leave this issue open for a while, see if somebody else has something interesting to say.

gijzelaerr, 2016-12-30:
issue has been open with 2 months of activity, I don't really know how to proceed so closing this issue. If you have an update please reopen.

closed 2016-12-30 by gijzelaerr

JKelly-Mevex, 2019-05-27:
I just had the same issue and found in the Snap7 Documentation that when doing a read_area:

**"If either S7AreaCT or S7AreaTM is selected, WordLen must be either S7WLCounter orS7WLTimer."**

I had made a change to snap7/Client.py and tested that it is now reading timers.  I currently have a Pull Request in to merge my changes into the master branch. Hope this helps.