gijzelaerr/python-snap7 #63: myplc = snap7.client.Client() error

pllow3, 2016-11-11:
I had try until step:
myplc = snap7.client.Client() got error syntax (Pls refer to attachment pic) TQ
[image]

gijzelaerr, 2016-11-11:
it looks like the snap7 dll is not in your path. http://python-snap7.readthedocs.io/en/latest/installation.html#windows

spreeker, 2016-11-11:
You did not install the snap7 libary correctly. Ease follow the installatie I struction. Python snap7 depends on the snap7 package.

spreeker, 2016-11-11:
You did not install the snap7 libary correctly. Please follow the installations structions... Python snap7 depends on the extrrnal (not Python) snap7 package.

closed 2016-11-11 by spreeker

pllow3, 2016-11-11:
I am using Raspberry pi3 python for testing. I install base on step:

1) sudo make –f arm_v7_linux.mk install (this step completed normal)

Can you advise what is the step needed? TQ

pllow3, 2016-11-11:
I am using Raspberry pi3 for testing. I install base on step:

1) sudo make –f arm_v7_linux.mk install (it install complete normal)
2) python setup.py install ( unit last line got error, it mention no permission to copy file into usr/local/lib/python2.7

Do u have this problem b4?

TQ