gijzelaerr/python-snap7 #64: test_client.py fail

binmulyadin, 2016-11-12:
I am new to python, github and all this stuff but used to work with PLC S7. I am using win7 64 bit, and running a virtual win xp as my virtual PLC. I have install snap7.dll and snap7 in "Windows/system32/ "folder and i think from "snap7.dll" to my virtual PLC is connected. I have test using clientdemo.exe file in his "rich demo folder" and it works.

Now I want to test with Python34 (I have find somewhere in the web that python-snap7 work with this version). What I have done is:
1. install with "pip install python-snap7"
2. open text_client.py, edit ip, rack and slot then run

I get the below error, please advice 

E.
======================================================================
ERROR: setUpClass (__main__.TestClient)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\Temp\python-snap7-0.5\python-snap7-0.5\test\test_client.py", line 30, in setUpClass
    cls.server_pid = Popen([server_path]).pid
  File "C:\Python34\lib\subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "C:\Python34\lib\subprocess.py", line 1112, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified

----------------------------------------------------------------------
Ran 1 test in 0.011s

FAILED (errors=1)

spreeker, 2016-11-12:
Read the example code that's out there. You seem to try to run the test code without the test server.

May I suggest you read some data from a test Plc?

spreeker, 2016-11-12:
Or Python just  cannot find the snap package.

binmulyadin, 2016-11-13:
Thanks to the author and thanks to spreeker for quick reply. Let me review again my situation:
1. I have Install (copy) snap7.dll and snap7.lib
2. Testing the connection from snap7 to PLC is OK (test using clientDemo.exe from snap7 in rich-demo folder)
now is the python part:
3. I have download and install python-snap7 (using setup.py it run without error)
4. Now then I open test_client.py (from test folder) and I got those above error (I change the ip, rack and slot accordingly in this file).

Now how to know that python is find the snap package? I test by writing (in python shell)
>>>import snap7
>>>
and no error

regards

binmulyadin, 2016-11-21:
Thanks all,
it works now, nothings wrong with snap7.dll and also python-snap7 installation. After testing from the "example" folder by typing line by line in python sell is a bit more clear.

closed 2016-11-21 by binmulyadin