gijzelaerr/python-snap7 #65: util.get_int() broken on Linux / Python 2.7.3

t-siedenburg, 2017-01-20:
When sending a Logo8 counter-value via network-analog-output to a snap7-server
the values returned from  util.get_int() were not always correct.
The problem is also reproducible with util.set_int() and util.get_int() :

```
import snap7
server = snap7.server.Server(log=True)
DB1 = (snap7.snap7types.wordlen_to_ctypes[snap7.snap7types.S7WLByte]*4)()
server.register_area(snap7.snap7types.srvAreaDB, 1, DB1)
for Data_Set in (-32768, -16385, -256, -128, -127, 0, 127, 128, 255, 256, 16384, 32767):
    snap7.util.set_int(DB1, 0, Data_Set)
    Data_Get = snap7.util.get_int(DB1, 0)
    print "%6d %6d %s" % (Data_Set, Data_Get, (Data_Set==Data_Get))

Output:

 -32768 -32768 True
 -16385 -16641 False
   -256   -256 True
   -128   -384 False
   -127   -383 False
      0      0 True
    127    127 True
    128   -128 False
    255     -1 False
    256    256 True
  16384  16384 True
  32767  32511 False
```

The problem is due to python sign handling of integers.
A working fix in util.get_int():
```
...
    byte1 = _bytearray[byte_index + 1] & 0xFF

Output:

-32768 -32768 True
-16385 -16385 True
  -256   -256 True
  -128   -128 True
  -127   -127 True
     0      0 True
   127    127 True
   128    128 True
   255    255 True
   256    256 True
 16384  16384 True
 32767  32767 True
```

byte0 is left with sign, because Logo int ranges from -32768 ... +32767

gijzelaerr, 2017-02-08:
hi @t-siedenburg , can you maybe issue a pull request with a unittest? That would simplify my life a lot.

gijzelaerr, 2017-03-02:
this issue has been solved in PR #71 ! will push out a release

closed 2017-03-02 by gijzelaerr