gijzelaerr/python-snap7 #66: can't find snap7 library. If installed, try running ldconfig

mostafamahmodi, 2017-02-05:
hi everyone 
i installed snap7 in virtualenv in ubuntu desktop by using pip installation
but when i run my program i get this error:
snap7.snap7exceptions.Snap7Exception: can't find snap7 library. If installed, try running ldconfig

i add snap7 package in os.path array:

`import argparse
import ConfigParser
import sys
import re
import os

sys.path.append('/home/mostafa/plc/local/lib/python2.7/site-packages/')
sys.path.append('/home/mostafa/plc/lib/python2.7/site-packages/')

from snap7 import client

parser = argparse.ArgumentParser()
group = parser.add_mutually_exclusive_group()
group1 = parser.add_mutually_exclusive_group()
config = ConfigParser.ConfigParser()
group1.add_argument("-r", help="read from config file", action="store_true", )
group1.add_argument("-s", "--specify", help='specify PLC IP address. example:"ip,port,slot"')
group.add_argument("-S", "--stop", help="stop PLC.", action="store_true")`

by the way i use pycharm IDE 
thank u

gijzelaerr, 2017-02-06:
did you run ldconfig?

gijzelaerr, 2017-02-08:
you probably didn't install the snap7 binary package. Please read the documentation

https://launchpad.net/~gijzelaar/+archive/ubuntu/snap7

closed 2017-02-08 by gijzelaerr

mostafamahmodi, 2017-02-10:
it is fixed thanks for all your advices

jst0319, 2019-12-12:
@gijzelaerr I am facing the same error. I have installed the required packages properly. Please help