gijzelaerr/python-snap7 #68: Client problem

Youandi12, 2017-02-20:
We get this error from Python and don't know how to solve it.
We want to use the Python to write values in a DB(109). They are real.
We have no experience with the raspberry, so could you help us please?

Traceback (most recent call last):
  File "/home/pi/testcom2.py", line 17, in <module>
    client = snap7.client.Client()
  File "/usr/local/lib/python2.7/dist-packages/snap7/client.py", line 36, in __init__
    self.create()
  File "/usr/local/lib/python2.7/dist-packages/snap7/client.py", line 43, in create
    self.library.Cli_Create.restype = c_void_p
AttributeError: 'NoneType' object has no attribute 'Cli_Create'
>>> 

This is our programme:

import re
from ctypes import c_int, c_char_p, byref, sizeof, c_uint16, c_int32, c_byte
from ctypes import c_void_p

import logging

import snap7
from snap7 import six
from snap7.snap7types import S7Object, buffer_type, buffer_size, BlocksList
from snap7.snap7types import TS7BlockInfo, param_types, cpu_statuses

from snap7.common import check_error, load_library, ipv4
from snap7.snap7exceptions import Snap7Exception

logger = logging.getLogger(__name__)

client = snap7.client.Client()
client.connect('192.168.5.2', 0, 1)

class Client(object):
    def db_write(self,db_number , start, data):
        """
        Writes to a DB object.

        :param start: write offset
        :param data: bytearray
        """
        db_number = 109
        start = 0
        data = ff
        

        
        wordlen = snap7.snap7types.S7WLReal
        type_ = snap7.snap7types.wordlen_to_ctypes[wordlen]
        size = len(data)
        cdata = (type_ * size).from_buffer(data)
        logger.debug("db_write db_number:%s start:%s size:%s data:%s" %
                     (db_number, start, size, data))
        return self.library.Cli_DBWrite(self.pointer, db_number, start, size,
                                        byref(cdata))

gijzelaerr, 2017-02-20:
It looks like that error can't come from the code you copy pasted. Also I get the idea are not really familiar with Python since you try to override the existing client code? I think it is wise to maybe follow some python tutorials.

Try something like this:

```
import snap 7
client = snap7.client.Client()
client.connect('192.168.5.2', 0, 1)
db_number = 109
start = 0
data = ff
client.db_write(self, db_number, start, data)
```

hope that gets you a bit further.

closed 2017-02-20 by gijzelaerr

Youandi12, 2017-02-20:
We have run your program and have the same problems  

Python 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2] on linux2
Type "copyright", "credits" or "license()" for more information.
>>> ================================ RESTART ================================
>>> 

Traceback (most recent call last):
  File "/home/pi/testcom3.py", line 2, in <module>
    client = snap7.client.Client()
  File "/usr/local/lib/python2.7/dist-packages/snap7/client.py", line 36, in __init__
    self.create()
  File "/usr/local/lib/python2.7/dist-packages/snap7/client.py", line 43, in create
    self.library.Cli_Create.restype = c_void_p
AttributeError: 'NoneType' object has no attribute 'Cli_Create'
>>>

gijzelaerr, 2017-02-20:
you don't get an other error about snap7 library not found? How did you install snap7? Did you modify the python-snap7 source code?

Youandi12, 2017-02-20:
These are the only errors we get. Normally the snap7 library is ok.
We have installed the snap7 library with: 
$ p7zip -d snap7-full-1.0.0.7z 
$ cd build/unix
$ make -f arm_v7_linux.mk     
pip install python-snap7

We didn't modify anything in the source code.

gijzelaerr, 2017-02-20:
what happens if you run
```
from ctypes.util import find_library
print find_library('snap7')
```
in a python console or script?

Youandi12, 2017-02-20:
we got this:
>>> 
libsnap7.so
>>>

gijzelaerr, 2017-02-20:
And what does this give you:
```
from ctypes import cdll
from ctypes.util import find_library
print cdll.LoadLibrary(find_library('snap7'))
```

Youandi12, 2017-02-20:
>>> 
<CDLL 'libsnap7.so', handle 5de320 at 756d0ff0>
>>>

gijzelaerr, 2017-02-20:
are you sure you didn't modify the client code? This should just work... try uninstalling python-snap7 and reinstalling from pip.

gijzelaerr, 2017-02-20:
hm you might have spotted a bug actually.

https://github.com/gijzelaerr/python-snap7/blob/master/snap7/common.py#L42

are you running this in an interactive terminal? it looks like if you called the find_library function before it will return None. Can you restart the interpreter?

gijzelaerr, 2017-02-20:
nah that doesn't seem the case.... hm

Youandi12, 2017-02-20:
we have changed the common.py en we add the lib_location='/usr/local/lib/libsnap7.so 
above return your marked return (boven return heb ik deze regel ingezet )

i have follow this 
You will need to edit the lib_location on common.py in the /usr/local/lib/python2.7/dist-packages/snap7/ directory
Add a line in the __init__ part of the Snap7Library class:
lib_location='/usr/local/lib/libsnap7.so' 
example below:

class Snap7Library(object):
    """
    Snap7 loader and encapsulator. We make this a singleton to make
    sure the library is loaded only once.
    """
    _instance = None
    def __new__(cls, *args, **kwargs):
        if not cls._instance:
            cls._instance = object.__new__(cls)
            cls._instance.lib_location = None
            cls._instance.cdll = None
        return cls._instance

    def __init__(self, lib_location=None):
        lib_location='/usr/local/lib/libsnap7.so' # add this line here
        if self.cdll:
            return
        self.lib_location = lib_location or self.lib_location or find_library('snap7')
        if not self.lib_location:
            msg = "can't find snap7 library. If installed, try running ldconfig"
            raise Snap7Exception(msg)
        self.cdll = cdll.LoadLibrary(self.lib_location)

Youandi12, 2017-02-21:
hi ,
We have reinstalled snap 7 and his library. We tested this like you told us yesterday and it says that the library is installed.
Our faults of yesterday are gone, but we have a new problem if we try this:
import snap 7
client = snap7.client.Client()
client.connect('192.168.5.2', 0, 1)
db_number = 109
start = 0
data = ff
client.db_write(self, db_number, start, data)

We get this fault:
Traceback (most recent call last):
  File "/home/pi/TEST3.py", line 8, in <module>
    client.db_write(self, db_number, start, data)
NameError: name 'self' is not defined
>>>

gijzelaerr, 2017-02-21:
don't give self to the method. Please google for 'python' and 'self', for example this explains more:

http://stackoverflow.com/questions/7721920/when-do-you-use-self-in-python

```
import snap 7
client = snap7.client.Client()
client.connect('192.168.5.2', 0, 1)
db_number = 109
start = 0
data = ff
client.db_write(db_number, start, data)
```

Youandi12, 2017-02-21:
if we remove self then we got this problems 
Traceback (most recent call last):
  File "/home/pi/TEST3.py", line 8, in <module>
    client.db_write( db_number, start, data)
  File "/usr/local/lib/python2.7/dist-packages/snap7/client.py", line 24, in f
    code = func(*args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/snap7/client.py", line 152, in db_write
    size = len(data)
TypeError: object of type 'int' has no len()

>>>

gijzelaerr, 2017-02-21:
i'm sorry but you really should get yourself more familiar with Python, otherwise this is going nowhere. Data should be a `bytearray`.

Zebrafish007, 2017-12-31:
It should be `data = 'ff'` # for all who just new to python.

Ankit9721, 2025-04-15:
gooood