gijzelaerr/python-snap7 #78: snap7 module not found windows 10

vrindakm, 2017-04-04:
Hello

I am a new to python (and to GitHub) and am trying it out for an automation project of mine.
I am trying to use 64-bit Python 3.5.3 with 64 bit Snap7 on Windows 10 (also 64 bit).
I have copied the .dll and .lib files of snap7 and added it to system32 folder of windows.
I have also copied it to my python path.
But, when I import snap7 , it says 'Importerror: no module named snap7'
Does python-snap7 work in Windows 10? The document said it 'may' work on other OS.
Pardon me if this is trivial. It would be very helpful if I can get some guidance.

Thank you.

gijzelaerr, 2017-04-04:
it should work. Maybe you need to modify the security of the dll? Maybe downloaded DLL's are not trusted by default on windows 10. Right click on the DLL in your system32 folder and somewhere there it should give an option to enable "trust this dll downloaded from a different computer".

Otherwise, make sure if system32 is in your PATH, not sure if that is the case on 64 windows? 

Sorry, can't think of much else here, hope this helps.

spreeker, 2017-04-04:
Switch to Ubuntu :)

vrindakm, 2017-04-04:
@gijzelaerr : I did copy the files as admin..so there seems to be no issue with security and yes, the system32 is indeed in my PATH 
Thanks for replying.

@spreeker : unfortunately, two main softwares of my project are not compatible with ubuntu :)
Thanks for replying though.

gijzelaerr, 2017-04-04:
check this. 
[image]
otherwise make sure you have the right architecture, not the 32 bit one. I can't really help you much further than this.

vrindakm, 2017-04-04:
I did. But I do not have this option at all. All I got was the admin permission request while copying them earlier.
[image]

I shall go through all of it again.Thanks for the help so far :)

closed 2017-04-04 by vrindakm