gijzelaerr/python-snap7 #79: Issue in "list_blocks_of_type code"

objectivewerks, 2017-06-30:
I was trying to get the "list_blocks_of_type" code working but every time I called it, I was returned a 10 Integer Array containing some rather large numbers.  After doing some debugging I deduced that if I took the large number in each array element and split it into 2 * 16 bit words then the array contained *some* of the block numbers of that block type.  I had previously authored some C++ code that used the Snap7 library and had this running so looked at my previous code and noticed that I was passing in (by reference) a TS7BlocksOfType structure which actually equated to a integer array that is 0x2000 in size.  I changed the line in client.py for the function "def list_blocks_of_type(self, blocktype, size):" that was

data = (c_int * 10)() to be data = (c_uint16 * 0x2000)()

I am now return an array of integers containing all of the data (and then some as the array is 0x2000 whether it is used or not). Perhaps the proper way to handle this is to use the "size" variable in the declaration of the data array

ie data = (c_uint16 * size)()

Then an array of integers the exact size required is returned.

gijzelaerr, 2017-07-06:
hi @objectivewerks thanks for your bug report. Do you think can do a PR with your fix and bundle it with a test to verify it works correctly? That would be a great help!

objectivewerks, 2017-09-09:
Hi,

Sorry for the long delay. Somehow this ended up in my spam folder. My
apologies.
Did you still want me to do a PR?

Let me know

James

gijzelaerr, 2017-09-15:
i think the problem is still there right? so a bug fix PR would be awesome :)

gijzelaerr, 2018-08-02:
this issue has already been resolved https://github.com/gijzelaerr/python-snap7/blame/master/snap7/client.py#L316

closed 2018-08-02 by gijzelaerr