gijzelaerr/python-snap7 #94: Writing to an INT memory, overwrites the memories below it

ildomarcarvalho, 2017-12-21:
Dear all,
When I try to write to an integer memory, it overwrites some memories below this int memory. Can you help me to find out if it is a bug or I just using the the snap7 incorrectly?

```python
import snap7

layout = """
      0       DB_VAR_1                    INT
      2       DB_VAR_2                    INT
      4.0     IN_Reserva_Camera           BOOL
      4.1     IN_Sinal_Iluminacao         BOOL
      4.2     IN_Robo_Trigger_Cam         BOOL
      4.3     IN_Robo_UR5_Inspecao        BOOL
      4.4     IN_Robo_UR5_Operacao        BOOL
      4.5     IN_Robo_UR5_Sinal_4         BOOL
      4.6     IN_Robo_UR5_Iniciar_ACK     BOOL
      4.7     IN_Reserva_2                BOOL
      5.0     IN_Sensor_Pressostato       BOOL
      5.1     IN_Reserva_Sensor_1         BOOL
      5.2     IN_Mon_BT_Emer_Painel_Pr    BOOL
      5.3     IN_Mon_BT_Emer_Reserva_1    BOOL
      5.4     IN_Mon_BT_Emer_Reserva_2    BOOL
      5.5     IN_Mon_Rele_BT_Emer         BOOL
      5.6     IN_Mon_Rele_Chave_Porta     BOOL
      5.7     IN_Reserva_3                BOOL
      6.0     IN_Sinal_MGB_Pos_Porta      BOOL
      6.1     IN_Sinal_MGB_Pos_Bolt       BOOL
      6.2     IN_Sinal_MGB_Pos_Trava      BOOL
      6.3     IN_Sinal_MGB_Diag_Saida     BOOL
      6.4     IN_Solic_Abert_Porta_MGB    BOOL
      6.5     IN_Sinal_Reset_MGB          BOOL
      6.6     IN_Reserva_4                BOOL
      6.7     IN_Reserva_5                BOOL
      7.0     IN_BT_Lig_Deslig_Comando    BOOL
      7.1     IN_BT_Pos_Inicial           BOOL
      7.2     IN_BT_Reset_Falha           BOOL
      7.3     IN_Cha_Manual_Automatico    BOOL
      7.4     IN_Reserva_6                BOOL
      7.5     IN_Reserva_7                BOOL
      7.6     IN_Reserva_8                BOOL
      7.7     IN_Reserva_9                BOOL
      8.0     IN_M_D_Ent_Trafo_Ant_Sec    BOOL
      8.1     IN_M_D_Sai_Trafo_Ant_Sec    BOOL
      8.2     IN_M_D_Ent_Med_Energia      BOOL
      8.3     IN_M_D_Vent_Painel          BOOL
      8.4     IN_M_D_Fonte_POE_Cam        BOOL
      8.5     IN_M_D_Controlador_UR5      BOOL
      8.6     IN_M_D_Luminaria_Painel     BOOL
      8.7     IN_M_D_Medidor_Energia      BOOL
      9.0     IN_M_D_Tomada_Manut         BOOL
      9.1     IN_M_D_Comando_Seguranca    BOOL
      9.2     IN_Reserva_10               BOOL
      9.3     IN_Reserva_11               BOOL
      9.4     IN_Reserva_12               BOOL
      9.5     IN_Reserva_13               BOOL
      9.6     IN_Reserva_14               BOOL
      9.7     IN_Reserva_15               BOOL
      10.0    IN_Sens_Cil_Index_Avanc     BOOL
      10.1    IN_Sens_Presenca_Prod_1     BOOL
      10.2    IN_Sens_Setup_2             BOOL
      10.3    IN_Sens_Setup_3             BOOL
      10.4    IN_Sens_Distrib_Res_1       BOOL
      10.5    IN_Sens_Distrib_Res_2       BOOL
      10.6    IN_Sens_Distrib_Res_3       BOOL
      10.7    IN_Sens_Distrib_Res_4       BOOL
      11.0    IN_Sens_Cil_Index_Recuad    BOOL
      11.1    IN_Sens_Presenca_Prod_2     BOOL
      11.2    IN_Sens_Setup_1             BOOL
      11.3    IN_Sens_Setup_4             BOOL
      11.4    IN_Sens_Distrib_Res_5       BOOL
[... 163 lines cut ...]
# This piece of code works without any problem
db_16[0].read(plc)
db_16[0]['IHM_Resultado_Insp_1'] = True
db_16[0].write(plc)

# If I try to write to this int memory, I end up writing to the memories
# IHM_Resultado_Insp_1, IHM_Gap_1 and IHM_Numero_Insp_2. Depending of the
# value, even more memories get overwriten (the IHM_Numero_Insp_2 memory gets
# the IHM_Numero_Insp_1 old value.
# The memory IHM_Numero_Insp_1 gets the right value
db_16[0].read(plc)
db_16[0]['IHM_Numero_Insp_1'] = 10
db_16[0].write(plc)

# I made this workaround to write correctly to int memories
def to_bytes(n, length, endianess='big'):
    h = '%x' % n
    s = ('0'*(len(h) % 2) + h).zfill(length*2).decode('hex')
    return s if endianess == 'big' else s[::-1]

IHM_Numero_Insp_1 = int(db_16[0]._specification['IHM_Numero_Insp_1'][0])
plc.db_write(db_number, IHM_Numero_Insp_1, to_bytes(0, 2))
```

Best Regards,

spreeker, 2017-12-21:
This could be a bug.

This code was used to read out 1000s of sensors and valves and to write
Boolean values. (On/off). Writing int values.might not have been properly
tested!!

There is no code dealing with the endian Ness (I did not need it at the
time)
 and now I see the 'to_int' function has no size check on it, I think its
current implementation is bad It just adds data to the the array and moving
everything if what is inserted is bigger than 2 bytes..leading to very
weird results! (Just tested it in a Python shell) a[2:4] = arraylagerthan2
just 'works' it just inserts everything starting at 2. Oops.

https://github.com/gijzelaerr/python-snap7/blob/master/snap7/util.py#L131

It should be easy for you to fix the offending function 'to_int' and do a
pull request.!

I do not have a PLC lying around anymore to test it properly.

Cheers!

Stephan.

ildomarcarvalho, 2018-01-03:
Thanks for your reply @spreeker 

I'm sorry I took this long to read your reply, but I posted this just the day I went on vacation.
Unfortunately I don't have a PLC with me anymore too. Maybe in a few weeks I'll have it again, then I will try to fix. If someone reading this post have a PLC, please feel free to fix :)

Thanks!

ildomarcarvalho, 2018-01-04:
I got my hands in a PLC today, so I could fix and test this bug. I just opened a pull request.

Thanks for the help

gijzelaerr, 2018-01-27:
I think this issue is solved by accepting #96. Please reopen if you think otherwise :)

closed 2018-01-27 by gijzelaerr