gijzelaerr/python-snap7 #95: missing function db_upload

Zebrafish007, 2017-12-30:
Hi,

I'm missing in client.py the function "db_upload" as addressed by function `make_item_db(dbnumber)` in example.py.

When changed `all_data = client.db_upload(db_number)` into `all_data = client.upload(db_number)` the following error occurs:

  >File "c:\python\anac2\lib\site-packages\snap7\client.py", line 194, in upload
    check_error(result, context="client")
  File "c:\python\anac2\lib\site-packages\snap7\common.py", line 65, in check_error
    raise Snap7Exception(error)
snap7.snap7exceptions.Snap7Exception: CPU : Function not available

What to do next?

gijzelaerr, 2017-12-30:
this is not an error in the python API. I guess the problem is what the error says the 'function is not availble' for your PLC.

Zebrafish007, 2017-12-31:
I've got the luxury of testing on S7-1212C AC/DC/RLY V4.0/V4.1.3 (Q3 2016) and on S7-1214C DC\DC\DC V4.0/V4.1.3 (Q1 2017) so I guess it should not give any problem from the Siemens side.  I'm also reading up all relevant closed bug reports here to solve this. Got the following checked for you:

python (2.7.14)
win10 x64 sp1
anaconda 4.3.30

    from ctypes import cdll
    from ctypes.util import find_library
    print cdll.LoadLibrary(find_library('snap7'))

Result:

<CDLL 'C:\Python\Anac2\snap7.dll', handle 9bc70000 at 4cc8908>

gijzelaerr, 2017-12-31:
python-snap7 is just a thin wrapper around the snap7 library. It is the PLC or the snap7 library raising this issue, not the python code. So please check the documentation of your PLC or open an issue at the snap7 issue tracker. I'm closing this issue because it seems to be unrelated to python-snap7. Please reopen if you think otherwise.

closed 2017-12-31 by gijzelaerr

Zebrafish007, 2017-12-31:
Created bugreport [here](https://sourceforge.net/p/snap7/discussion/bugfix/thread/e36b98c8/) on sourceforge.net.

gijzelaerr, 2017-12-31:
That is a different error. In that issue you report a `CPU : Item not available`, while here you report ` CPU : Function not available`.

Zebrafish007, 2017-12-31:
Very sharp! I'll check again. Also busy with getting a rebuild using vs2017 community in vs2015 mode with the ms sdk kit2017. Rebooting..back in a sec trying to solve this and the other error with reconstructed dll.

Zebrafish007, 2017-12-31:
With db_number = 109 the error is: snap7.snap7exceptions.Snap7Exception: CPU : Item not available

With db_number = 1 the error is:  snap7.snap7exceptions.Snap7Exception: CPU : Address out of range

With Tia portal v13 I've made DB1 having name: "Data_block_1". There is created at row 2 the variable with following info: 
name = hello, 
data type = bool,
start value = true, 
settings: retain/setpoint = unchecked,Accessebile/visible from HMI = checked.

I reckon its the `data = 'ff'` throuwing the error? if so what should I put there.. with tia portal v13 I've made above db1.

script code:

client = snap7.client.Client()
        client.connect('192.168.200.15', 0, 1)
        db_number = 1 # or 109
        start = 0
        data = 'ff'    # if ff is not inside '' it throws a linkage error.
        client.db_write(db_number, start, data)
        print client.db_read(self, db_number, start)

spreeker, 2017-12-31:
I would suggest you look at the example code. You should write a bytearray
in the db memory. Python  types  should be converted to types for the plc.
A bool variable could be in any of the 7 locations within a bool bytearray.
For example 0000001 or 0001000 etc. You need to look in the db details in
the Tia portal .

Writing 1111111 , to get true actualy means you write true on 7 possible
Boolean locations .

Cheers,

Stephan.

Zebrafish007, 2017-12-31:
The example.py code is what 'm testing and the stuff mentioned in the bug reports to get me on the road. But in both cases I hit a wall of bricks. 

code that  works (except for print 4 "MW20") stuff:

        print '1  %s    : %s' % (adress[0], plc.getMem('MX0.1')) # read memory bit M0.1
        print '2  %s    : %s' % (adress[1], plc.getMem('IX0.0')) # read input bit I0.0
        plc.writeMem('FREAL10',3.1415927)
    #    plc.writeMem('FREAL10',1.0001)
        print '3A %s : %s' % (adress[2][0], plc.getMem("FREAL100")) # read real from MD100
        print '3B %s  : %s' % (adress[2][1], plc.getMem("FREAL10")) # read real from MD10
    #    print '4 %s : %s' % (adress[3], plc.getMem("MW20")) # read int word from MW20
        print '5  %s     : %s' % (adress[4], plc.getMem("MB24", 254)) # write to MB24 the value 254
        print '\n'

The error:   File "D:\Scripts\ZFES2_0\S7_1212C.py", line 73, in getMem
    return get_int(mbyte,start)
  File "c:\python\anac2\lib\site-packages\snap7\util.py", line 148, in get_int
    value = struct.unpack('>h', struct.pack('2B', *data))[0]
struct.error: pack expected 2 items for packing (got 0)

So I disabled print4.

Then I go to just read/write just one thing to a DB1.. and also that fails. First due to `db_upload` (should be `upload` due to change in def name. Then due to  unknown things (see above)... then reinstall python-snap ..again via pip/delete snap7.dll and place new one from zip back... and again failure. As Gijs was very sharp two different error type... Now again stuck on that memmory thing...

The only referral is on the snap7 sourceforge under #Sharp7. Here is only explained the general addressing at the ReadArea table Area table... etc. So, I'm far from a PLC gury ..and this memory stuff is still yibberish to me.

I just want to read/write any value from a variable I've made in DB1. How simple can it be?

Zebrafish007, 2017-12-31:
Don't get me wrong... I'm delighted 100% that you guys did put so much work into this wrapper and snap7 to get it somehow on some machine working!

spreeker, 2017-12-31:
It's not so hard.

You can find out the exact memory layout of a db with the Siemens software.

Reinstalling software is really not needed.

The example code only works with a PLC with the exact same memory layout
which is not the case / the same for any / your PLC.

So you really do need to figure out which bits and bytes to read and write
in your PLC.

You can find out  the location of your created variables in the db by
looking what the Siemens software created for you.

The start of data could very well not be the first byte. 0. But for example
the 4th.

7 Booleans can be put in 1 byte.  The addressing is the incase of byte 0 ->
0.0, 0.1, 0.2, 0.3, 0,4, 0.5, 0,6 , 0.7.

For an int it is often 2 bytes.

So it is crucial you figure out the memory locations of the variables you
want to read / write.

Stephan.

Zebrafish007, 2017-12-31:
If you refer to Q0.0-0.7 or I0.0-0.7 in and outlets... that works fine. If I'd do Q12.7 I refer to an SM1231 port 8. That stuff works nicely (which enables me to recode all progs in python and skipp all stuff I've put into the PLC..but optical sensors with AI output is stuff I'd like to handle in the PLC and not on the PC. So.. this database DB1 is made with "load mem" 961 bytes and work mem 108 bytes.. somewhere in to PLC. The DB name is "Data_blovk_1", variable name is "Var1",type =  bool, value = true.

Whatdo I nedd to do:

    client = snap7.client.Client()
    client.connect('192.168.200.15', 0, 1)
    db_number = 1
    start = 0  # is this value correct?
    data = 'x' # what goes here 
    client.db_write(db_number, start, data)
   print client.db_read(self, db_number, start)

spreeker, 2017-12-31:
https://github.com/spreeker/python-snap7/blob/master/example/db1_layout.txt

Here you see  an example db.layout.
The user variable in this case start byte 4.

This layout / information is what you need to get from the Siemens
software.

Also studying
https://github.com/spreeker/python-snap7/blob/master/snap7/util.py

Will help your understanding!

Stephan.

Zebrafish007, 2017-12-31:
When I run example.py I hit the wall make_item_db at `all_data = client.db_upload(db_number)` mentioned above. If `all_data = client.db_upload(db_number)` changed into `all_data = client.upload(db_number)` I hit the wall: snap7.snap7exceptions.Snap7Exception: CPU : Function not available... code: <CDLL 'C:\Python\Anac2\snap7.dll', handle 98ba0000 at 5bfef60>

When I try db_read... I hit above mentioned wall... Thus.. you send me on an infinity loop I can't get out from...

spreeker, 2017-12-31:
I think db_uploads needs (less) different parameters then upload(). The
first function is sending a specific db number from PLC to the client. And
'upload' needs an extra memory type as argument Of what kind of memory
object you would like to have. Db. M, Fb, Ob, etc etc.

Upload_db (1) is equal to upload( type.Db, 1)

If I remember correctly....

Zebrafish007, 2018-01-01:
working on that tomorrow. I've made some progress in recompiling and informed Davide about my findings. Also the overlooking and mixing-up DB1 with OB1 kenobi. Now at` ctypes.ArgumentError: argument 2: <type 'exceptions.TypeError'>: Don't know how to convert parameter 2` but I think I'm able to fix that tomorrow with your latest comment.

JosephusMiller, 2018-09-03:
having got the same problem here `all_data = client.upload(6)` only brings an error. 
As I understood it's also possible to make a connection to the database throuhg snap7.util.DB(...) via a directly given "all_data". What type format would this exactly be? 

By the way I can easily read  with something like 'client.db_read(6,176,22)' from the database...

JosephusMiller, 2018-09-03:
Solved: I simply directly loaded the data `all_data = client.db_read(6,0,27)` - since I know from where to where my data goes: no problem.

Another thing: 
`db1[0].write()` should be `db1[0].write(client)`
Then everything works as expected. Nice. :)