gijzelaerr/python-snap7 #101: util.get_int() broken on OS X / Python 3.5

MAKOMO, 2018-02-19:
The example code of Issue #65 fails on OS X 10.13.3 (snap7 v1.4.2, python-snap7 v0.9) also due some sign handling confusion.

```
# python3 test.py
Traceback (most recent call last):
  File "test.py", line 8, in <module>
    Data_Get = snap7.util.get_int(DB1, 0)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/snap7/util.py", line 148, in get_int
    value = struct.unpack('>h', struct.pack('2B', *data))[0]
struct.error: ubyte format requires 0 <= number <= 255
```

while the test case that was added after fixing Issue #65 (as all other test_util tests) succeed

```
# cd test
# python3 test_util.py
.............
----------------------------------------------------------------------
Ran 13 tests in 0.004s

OK

```

The following shortend test2.py shows that the issue lies again in the encoding of integers larger than 127

```
import snap7
server = snap7.server.Server(log=True)
DB1 = (snap7.snap7types.wordlen_to_ctypes[snap7.snap7types.S7WLByte]*4)()
server.register_area(snap7.snap7types.srvAreaDB, 1, DB1)

for Data_Set in (0, 127, 128):
    snap7.util.set_int(DB1, 0, Data_Set)
    Data_Get = snap7.util.get_int(DB1, 0)
    print("%6d %6d %s" % (Data_Set, Data_Get, (Data_Set==Data_Get)))

# python3 test2.py
     0      0 True
   127    127 True
Traceback (most recent call last):
  File "test2.py", line 8, in <module>
    Data_Get = snap7.util.get_int(DB1, 0)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/site-packages/snap7/util.py", line 148, in get_int
    value = struct.unpack('>h', struct.pack('2B', *data))[0]
struct.error: ubyte format requires 0 <= number <= 255
```

Replacing the call `snap7.util.get_int(DB1, 0)` by the following modified `get_int(DB1, 0)` succeeds

```
import struct
def get_int(_bytearray, byte_index):
    """
    Get int value from bytearray.

    int are represented in two bytes
    """
    data = _bytearray[byte_index:byte_index + 2]
    data[1] = data[1] & 0xFF
    value = struct.unpack('>h', struct.pack('2B', *data))[0]
    return value

# python3 test2.py
     0      0 True
   127    127 True
   128    128 True
```

Accessing a server via TCP using a snap7 python client to read an integer via `snap7.util.get_int` that has been set on the server via `snap7.util.get_int` to a value larger than 127 works as expected.

gijzelaerr, 2018-02-20:
hi @MAKOMO thanks for your analysis! If this solves the problem, can you do a pull request? Thanks!

MAKOMO, 2018-02-20:
Dear @gijzelaerr, this hack solve the problem for me on this specific case. I have no idea of S7 nor snap7 nor python snap7. It is very likely that this hack breaks everything else. I hope that someone with some understanding of those tools and what exactly is going on here can resolve this correctly.

closed 2018-08-02 by gijzelaerr