gijzelaerr/python-snap7 #103: ERROR:snap7.common:b' TCP : Unreachable peer'

Tehnoves, 2018-03-01:
I'm trying to contact LOGO! 8. nothing happened. Where I'm wrong
[image]
[image]
[image]

openarchitect, 2018-03-01:
Hallo, a Short answer from my phone. If you have a TSAP 03.00 take 0x300 in connection call. TSAP 20.00 is 0x2000 address. Good luck

Gesendet mit meinem iPhone

Tehnoves, 2018-03-02:
I, like a sorcerer, danced with a tambourine around the fire, clockwise and counterclockwise, barked at the moon and howled at the stars, but alas ..
I tried everything I could think of a negative result
[image]

spreeker, 2018-03-02:
Looks like a network /  firewall issue?

openarchitect, 2018-03-02:
Met me Look at the Weekend about your issue.

Gesendet mit meinem iPhone

Tehnoves, 2018-03-03:
And why the Snap7 Clirnt 
Demo client does not have these problems. The network is directly IBM PC and LOGO! through a twisted cable.

[image]

closed 2018-03-03 by Tehnoves