gijzelaerr/python-snap7 #105: Logo 8, ValueError: invalid literal for int() with base 10

impulse3es, 2018-03-19:
I´ve been trying to read and write VM values following the Logo  example "logo_7_8.py", I´m using a logo 8, every time I try to access (read/write) a VM value bigger than 1000 i get an error "ValueError: invalid literal for int() with base 10", logo 8 VM values start at V1024.0.

[LOGO! 0BA8 Reserved VM addresses.pdf](https://github.com/gijzelaerr/python-snap7/files/1826927/LOGO.0BA8.Reserved.VM.addresses.pdf)

openarchitect, 2018-03-22:
There is an restriction for 3 digits in the read and write method. I will fix the error in the next days.
The regular expression allow only {1,3} and {1,4} is needed for Logo 8. Unfortunately I had only a Logo 7 for real testing.

gijzelaerr, 2018-03-26:
solved by PR #106

closed 2018-03-26 by gijzelaerr