gijzelaerr/python-snap7 #108: cannot import name 'util'

kavko8, 2018-04-04:
When I try to import snap7 I get this error:

Traceback (most recent call last):
  File "C:/Users/Matic/AppData/Roaming/Microsoft/Windows/Start Menu/Programs/Python 3.5/snap7.py", line 2, in <module>
    import snap7
  File "C:/Users/Matic/AppData/Roaming/Microsoft/Windows/Start Menu/Programs/Python 3.5\snap7.py", line 3, in <module>
    from snap7 import util
ImportError: cannot import name 'util'

Does anybody have any ideas?

kavko8, 2018-04-04:
Its "util" (from snap7 import util)

spreeker, 2018-04-05:
What is mean is this:

somotheomodule import util
from snap7 import util

kavko8, 2018-04-24:
I added the snap7.dll to the directory of Python installation folder and now it's working great.

closed 2018-04-24 by kavko8