gijzelaerr/python-snap7 #110: FreeBSD 11.1-RELEASE-p10

Ofloo, 2018-06-23:
I've installed python-snap7 through, ..

python3 -m pip install python-snap7
installed from git using 
python3 setup.py install

however both had the same result, this script works under linux ubuntu but not under FreeBSD 11, .. why is this?

# python3
Python 3.6.5 (default, May 10 2018, 01:16:34) 
[GCC 4.2.1 Compatible FreeBSD Clang 4.0.0 (tags/RELEASE_400/final 297347)] on freebsd11
Type "help", "copyright", "credits" or "license" for more information.
>>> import snap7
>>> 

# python3 test.py 
Traceback (most recent call last):
  File "test.py", line 10, in <module>
    plc = c.Client ()
  File "/usr/local/lib/python3.6/site-packages/snap7/client.py", line 35, in __init__
    self.library = load_library()
  File "/usr/local/lib/python3.6/site-packages/snap7/common.py", line 54, in load_library
    return Snap7Library(lib_location).cdll
  File "/usr/local/lib/python3.6/site-packages/snap7/common.py", line 46, in __init__
    raise Snap7Exception(msg)
snap7.snap7exceptions.Snap7Exception: can't find snap7 library. If installed, try running ldconfig
Exception ignored in: <bound method Client.__del__ of <snap7.client.Client object at 0x8007c9dd8>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/snap7/client.py", line 39, in __del__
    self.destroy()
  File "/usr/local/lib/python3.6/site-packages/snap7/client.py", line 54, in destroy
    return self.library.Cli_Destroy(byref(self.pointer))
AttributeError: 'Client' object has no attribute 'library'

gijzelaerr, 2018-06-25:
did you install the snap7 library itself?

Ofloo, 2018-06-25:
Yes I did it installed it self in /usr/lib I even made a link to /usr/local/lib

Installed the latest did gmake -f bsdmakefile, had to adjust g++ and cc to clang++ and clang  respectively. 

Regards,.. 

⁣Verzonden door ​⁣⁣BlueMail​​⁣

​

oor ​⁣BlueMail​⁣
​

oor BlueMail ​

gijzelaerr, 2018-06-25:
the problem is that the shared library can't be found:

```
 snap7.snap7exceptions.Snap7Exception: can't find snap7 library. If
⁣ installed, try running ldconfig
```

I don't have a freebsd machine available, but if you can't manually load
the shared library using ctypes then it is not a problem with python-snap7.

Ofloo, 2018-06-25:
Tried that as well. 

Regards,.. 

⁣Verzonden door BlueMail ​

gijzelaerr, 2018-06-25:
if you can't load the module that way there is a problem with your shared library loading setup.

gijzelaerr, 2018-08-02:
closing issue since reporter doesn't reply. the snap7 library doesn't load on his/her system.

closed 2018-08-02 by gijzelaerr