gijzelaerr/python-snap7 #111: get_connected returns true if the connection is broke

EnricGarrigo, 2018-07-12:


N3mesisXS, 2018-08-01:
I can confirm, had to use `If str(plc.get_cpu_state()) == "S7CpuStatusRun":`
Works like charm but feels ghetto. 

get_Connected() returns False when you run `plc.disconnect()`.

gijzelaerr, 2018-08-02:
since python-snap7 is a thin wrapper around snap7, this is actually a bug in the snap7 lib. So the long-term solution is to report it upstream. The only problem is that last time I checked the project is not really maintained anymore, so that fix might never happen.

I could add @N3mesisXS's suggestion to python-snap7, but I'm not sure if that works correctly for all other users. So @EnricGarrigo, for now, I suggest adding this check to your code and see if that solves the problem.

spreeker, 2018-08-02:
davide is doing his best to maintain it right? I think he will fix it if
you report it :)

gijzelaerr, 2018-08-02:
there have been no commits to the repo for more than a year...

@EnricGarrigo if you could report the issue upstream and link back here that would be useful.

spreeker, 2018-08-02:
Luckily PLC's are made to run for 25 years or so. :). Interfaces do not
change much. They have improved encryption on the newer models which are
slowly very slowly being deployed..

houssemELbahri, 2019-04-09:
evert time i try to connect my raspberry pi to a s7-1200 
i had this problem tcp time out error

DasLukas, 2022-11-06:
Hey everyone. Something new here? I need to check if the Logo reachable.

I wonder where to place @N3mesisXS workaround?

MountainKing91, 2022-11-20:
My workaround for this situation is to:
1) in the read loop I check the connection status before reading, if not connected I try to connect 
2) catch the read/write exception and disconnect the client. At that point get_connected will return false, so when checking the connection in the read loop (1) i will know I have to reconnect
This is also how Davide does in some examples (not in python).

BA7LYA, 2024-11-26:
I run into the same issue with @EnricGarrigo .