gijzelaerr/python-snap7 #117: client.get_plc_datetime() not working

Hennsi, 2018-11-02:
Hello guys, I am new here so sorry if my question is dumb. 

I am using Windows 7 running Python 3.7 and a CPU1517 connected over Ethernet. 
The connection works great and I am already able to read out PLC variables/ DB's. When I try to read the PLC datetime with the function client.get_plc_datetime() I get the following error:

```
Traceback (most recent call last):
  File "read_plc.py", line 41, in <module>
    print(client.get_plc_datetime())
  File "C:\Users\user1\AppData\Local\Programs\Python\Python37\Lib\site-packages\snap7\client.py", line 589, in get_plc_datetime
    check_error(result, context="client")
  File "C:\Users\user1\AppData\Local\Programs\Python\Python37\Lib\site-packages\snap7\common.py", line 65, in check_error
    raise Snap7Exception(error)
snap7.snap7exceptions.Snap7Exception: b'CPU : Item not available'
```
line 41 in read_plc.py: print(client.get_plc_datetime())

In client.py I only added:
`...
day = buffer[3] +1, ... `
in our get_plc_datetime() function

Thank you guys!

gijzelaerr, 2019-01-05:
I don't think this is a problem with python-snap7 but with your PLC not supporting this call or you not having enough permissions.

Hennsi, 2019-01-10:
Hello, thanks for your answer!
I did not solve the "get_plc_datetime()" problem yet but I solved my problem a different way.
Maybe somebody may need this information or has similar issues.
I configured a Datetime long(DTL) PLC in TIA V15 and I read it out like this before i read out the other PLC's:

```
def readtime(snap7data):
    year = util.get_int(snap7data,0)
    month = snap7data[2]
    day = snap7data[3]
    ####I did not need weekday####
    hour = snap7data[5]
    minute = snap7data[6]
    second = snap7data[7]
    microsecond = int(util.get_dword(snap7data,8)/1000) # Nanosecond(4 Bytes Byte 8-11)
    return datetime(year, month, day, hour, minute, second, microsecond)
```

By the way, thank you very much for your python-snap7!

closed 2019-02-15 by Hennsi