gijzelaerr/python-snap7 #126: Export Symbol Table

Adrian-at-CrimsonAuzre, 2019-07-23:
This isn't an issue with the snap7 library or anything, but I can't find any documentation so I'll ask here.
How do I get a text symbol table exported from Step 7 project for use in this library?
I see this in the examples but I want to know how to obtain it:
```
0       RC_IF_ID         INT
2       RC_IF_NAME       STRING[16]
20      BatchName        STRING[32]
54      BatchID          DWORD
58.0    Occupied         BOOL     # Bezet door Koole PA
60      Status           INT      # Status
62      SubStatus        INT      # Substatus
64      Opdrachtstatus   INT      # Opdrachtstatus (zie enum)
66      Tankstatus       INT      # Tankstatus     (zie enum)
68.0    RC_IF_ERR        BOOL     # Module Fout
68.1    QOk              BOOL     # Module OK Module specifieke parameters
68.2    Unlock           BOOL
70      Cmd              INT      # Commando
72      T_Uit            REAL     # Uitschakeltijd
76      Massa            REAL     # Massa product in tank
80      MaxMassa         REAL     # Maximale capaciteit van tank
84      T_Aflever        REAL     # Aflevertemperatuur
88      T_Max            REAL     # Maximale product temperatuur
92      T_HH             REAL     # Hoog Hoog alarm product temperatuur
96      T_LL             REAL     # Laag Laag alarm product temperatuur
100     T_Schakel        REAL     # Schakeltemperatuur productverwarming
104     T_Warmwater      REAL     # Maximale watertemperatuur verwarming
108     SG               REAL     # Soortelijk gewicht bij temperatuur SG_T
...
```
Also, since I'm not very familiar with PLCs, could I get a little more explanation on the [snap7.util.DB](https://python-snap7.readthedocs.io/en/latest/util.html#snap7.util.DB) function? What does the ```size``` field mean?

gijzelaerr, 2019-10-01:
Hi sorry, but I cant help you. This is the bug tracker (only) for the python-snap7 software.

closed 2019-10-01 by gijzelaerr

spreeker, 2019-10-02:
If you work in your siemens plc environment there should be a symbols mapping somewhere from which you can simply copy and paste the values.