gijzelaerr/python-snap7 #127: client.full_upload() return bytearray size 65536

Lamelynx, 2019-08-29:
Shouldn't client.full_upload() return the actual FC's bytearray size?
In my code i do it manually with:
return =  client.full_upload('FC', 1000)
client.download(result[0][:result[1])

If i download full result[0] bytearray i got size error.
Tested on S7-400 ans PLCsim

Lamelynx, 2019-08-29:
Just for clarification:
client.full_upload() return the actual size as result[1] but len(result[0]) returns 65536

spreeker, 2019-09-03:
Hai!  This could very well be the case.  Could you create a pull request with a test showing your solution?

Many function have never been used by me in production so still could contain mistakes or do not handle some edge conditions well. It could also be simple snap7 libary behaviour in which case we cannot do something about it here.

nikteliy, 2020-12-02:
I have tried to reproduce bug on my ET200S. No luck. Is it still a thing?
[image]

swamper123, 2020-12-02:
Since 2019 some changes have been made, which may have fixed it. (see #158 )
I am not able to verify that one, sorry. :/

gijzelaerr, 2020-12-02:
ah yes, forgot to close this issue I realise now.

closed 2020-12-02 by gijzelaerr