gijzelaerr/python-snap7 #130: read_multi.py in examples folder not returning any results

edsase, 2019-10-01:
Hi, thank you contributors for your wonderful library.
I have an issue with the read_multi.py script in the example folder.
I tried to reproduce it as much as possible but it is not getting any result.
I don't get any errors but the values returned are always 0. 
So it seems like the pdata /buffer is not being filled with values.
I can however read the same addresses using **read_area** function.
The code I used is an exact copy of read_multi.py with changes to the DB addresses.
Thanks for looking into this.

spreeker, 2019-10-02:
The examples read plc data, but using the example code will not directly
for *anyone* since the target PLC and data in that PLC is most likely
different.
So you really need to modify the examples to work with YOUR plc and data in
there.
There are plenty of reasons why your code is not working. post it in a gist?

Stephan.

edsase, 2019-10-02:
Hi Stephan, I understand that the example will not work for anyone except if it is modified accordingly. This is what I did. I am using an S7-319 CPU and trying to read values from DB1.  See comments in UPPERCASE on the values I wanted to read out. I can read these values correctly using the read_area function but not with the read_multi_vars. Here's is my code:

`import ctypes
import struct

import snap7
from snap7.common import check_error
from snap7.snap7types import S7DataItem, S7AreaDB, S7WLByte

client = snap7.client.Client()
client.connect('192.168.0.171', 0, 2) # CHANGED PLC IP ADDRESS

data_items = (S7DataItem * 3)()

data_items[0].Area = ctypes.c_int32(S7AreaDB)
data_items[0].WordLen = ctypes.c_int32(S7WLByte)
data_items[0].Result = ctypes.c_int32(0)
data_items[0].DBNumber = ctypes.c_int32(1) # CHANGED DB NUMBER
data_items[0].Start = ctypes.c_int32(0) # START OF DB1.DBB0
data_items[0].Amount = ctypes.c_int32(4)  # I PUT A REAL IN DB1.DBD0

data_items[1].Area = ctypes.c_int32(S7AreaDB)
data_items[1].WordLen = ctypes.c_int32(S7WLByte)
data_items[1].Result = ctypes.c_int32(0)
data_items[1].DBNumber = ctypes.c_int32(1)
data_items[1].Start = ctypes.c_int32(4) # START OF DB1.DBD4
data_items[1].Amount = ctypes.c_int32(4)  # I PUT A REAL IN DB1.DBD4

data_items[2].Area = ctypes.c_int32(S7AreaDB)
data_items[2].WordLen = ctypes.c_int32(S7WLByte)
data_items[2].Result = ctypes.c_int32(0)
data_items[2].DBNumber = ctypes.c_int32(1) # CHANGED DB NUMBER
data_items[2].Start = ctypes.c_int32(8) # START OF DB1.DBW8
data_items[2].Amount = ctypes.c_int32(2)  # I PUT A WORD IN DB1.DBW8

# create buffers to receive the data
# use the Amount attribute on each item to size the buffer
for di in data_items:
    # create the buffer
    buffer = ctypes.create_string_buffer(di.Amount)

    # cast the pointer to the buffer to the required type
    pBuffer = ctypes.cast(ctypes.pointer(buffer),
                          ctypes.POINTER(ctypes.c_uint8))
    di.pData = pBuffer

result, data_items = client.read_multi_vars(data_items)

for di in data_items:
    check_error(di.Result)

# struct formats
fmts = ['>f', '>f', '>h']

# unpack and print the result of each read
for i in range(0, len(data_items)):
    fmt = fmts[i]
    di = data_items[i]
    foo = ''.join([chr(di.pData[i]) for i in range(0, di.Amount)])
    fnum = struct.unpack(fmt, foo)[0]
    print(fnum)

client.disconnect()
client.destroy()`

Thanks for helping me resolve this.

Mvg,

Edmond

edsase, 2019-10-05:
Hi, I was able to find the error and  have submitted a pull request with the correction.

closed 2019-10-05 by edsase