gijzelaerr/python-snap7 #131: Unable to read any DB or variable from S7-1200

mikla90, 2019-10-04:
Hi. Whenever I try to read or write a variable in my S7-1200 I always get a  "Function refused by CPU" error. I'm able to connect to the PLC without issues, but everytime I try to read or write DBs it gets an error. The code is taken from here [http://simplyautomationized.blogspot.com/2016/02/python-snap7-s7-1200-simple-example.html](url)

and modified with the correct IP adress and variables, but I can't get it to work. I've already set up my PLC with the security instructions as per the original Snap7 instrucions.

Thanks

edsase, 2019-10-04:
Perhaps you may have tried this already but when I was in this same situation, it was because I did not **Uncheck Optimized block access** of the DB. This advice is indeed in the original Snap7 instructions but I missed it somehow:
1. Select the DB in the left pane under “Program blocks” and press Alt-Enter (or in the contextual menu select “Properties…”)
2. Uncheck Optimized block access, by default it’s checked.

edsase, 2019-10-04:
Not sure if it has to do with you windows version as to my knowledge, nothing is said about bit version on the docs. By the way, are you able to read other areas like inputs and markers that are not in DBs?

mikla90, 2019-10-04:
Nope, nothing. That's my main issue. By the way, I'm thinking right now that I can't find the Optimized block option in the outputs section. I've tried with other DBs created for the test, but no luck. Where is it for the outputs?

edsase, 2019-10-04:
Attached photo shows where you can find the optimized block access option.
For further troubleshooting, you may also want to use the Lazarus located in the snap7 folder: snap7-full-1.4.2\rich-demos\i386-win32\bin\clientdemo.exe

[optimzedblock.pdf](https://github.com/gijzelaerr/python-snap7/files/3692201/optimzedblock.pdf)

mikla90, 2019-10-04:
I know where to find that, thanks. I can't find it for the hardware inputs and outputs cause they're just in the variables table, not in a DB. Or at least that's my understanding, I'm quite new to Siemens.

edsase, 2019-10-04:
[This tutorial](http://simplyautomationized.blogspot.com/2016/02/python-snap7-s7-1200-simple-example.html) might be useful. Check it out.

mikla90, 2019-10-05:
That's exactly where I got my code. I even went and configured the same variables in the program, but no luck. I'm thinking that there's either a config error on the PLC or on my Windows machine.

mikla90, 2019-10-05:
So, I've ginally been able to retrieve my code. Below it is:

`import snap7.client as c
from snap7.util import *
from snap7.snap7types import *
import snap7
import time

'''
areas = ADict({
    'PE': 0x81,
    'PA': 0x82,
    'MK': 0x83,
    'DB': 0x84,
    'CT': 0x1C,
    'TM': 0x1D,
})
'''

def WriteMemory(plc, byte, bit, datatype, value):
    result = plc.read_area(areas['MK'], 0, byte, datatype)
    if datatype == S7WLBit:
        set_bool(result, 0, bit, value)
    elif datatype == S7WLByte or datatype == S7WLWord:
        set_int(result, 0, value)
    elif datatype == S7WLReal:
        set_real(result, 0, value)
    elif datatype == S7WLDWord:
        set_dword(result, 0, value)
    plc.write_area(areas["MK"], 0, byte, result)

def ReadMemory(plc, byte, bit, datatype):
    result = plc.read_area(areas['MK'], 0, byte, datatype)
    if datatype == S7WLBit:
        return get_bool(result, 0, bit)
    elif datatype == S7WLByte or datatype == S7WLWord:
        return get_int(result, 0)
    elif datatype == S7WLReal:
        return get_real(result, 0)
    elif datatype == S7WLDWord:
        return get_dword(result, 0)
    else:
        return None

if __name__ == "__main__":
    plc = c.Client()
    plc.connect('192.168.1.11', 0, 1)
    print(plc.get_connected())
    time.sleep(2)
    WriteMemory(plc, 100, 0, S7WLBit, 0)
    print('Done')
    print(ReadMemory(plc, 100, 0, S7WLWord))
`

One thing I noticed is that I get 61 "Unused import <function>from wildcard import" where I have 61 different functions. Is this of any help?

Thanks

gijzelaerr, 2020-04-16:
This issue didn't see any activity for a long time now and it seems that the original question has sort of been answered.

closed 2020-04-16 by gijzelaerr

hiteshpanchani, 2021-01-08:
DB read function is broken with latest firmware version of Siemens PLC. I am able to read Memory block and PE. Please see below error samples for more details, 

PLC - S7 1200 
Firmware Version  - 4.4 
CPU - 1215c

Error Sample 1:
`b'CPU : Address out of range'
Traceback (most recent call last):
  File "read_multi.py", line 39, in <module>
    check_error(di.Result)
  File "C:\Python\lib\site-packages\snap7\common.py", line 66, in check_error
    raise Snap7Exception(error)
snap7.exceptions.Snap7Exception: b'CPU : Address out of range'`

Error Sample 2:

`THIS IS EXAMPLE CODE MEANTH TO BE READ.

It is used to manipulate a large DB object with over
450 'rows' which represent valves

You don't have a project and PLC like I have which I used
to create the test code with.

Traceback (most recent call last):
  File "example.py", line 245, in <module>
    print_tag()
  File "example.py", line 229, in print_tag
    db1 = make_item_db(1)
  File "example.py", line 201, in make_item_db
    all_data = client.db_upload(db_number)
AttributeError: 'Client' object has no attribute 'db_upload'`

swamper123, 2021-01-08:
@hiteshpanchani
Your Example 2 (from example.py:

> `THIS IS EXAMPLE CODE MEANTH TO BE READ.

says it all..

For your Example 1, we may need more information.
- did the connection work?
- was it working with lower versions?
- are the plc values exposed like it is mentioned at http://snap7.sourceforge.net/ ?
- what is the PLC address of your value and how is your read command?