gijzelaerr/python-snap7 #143: dint compatibility

terryrankine, 2020-01-08:
Hi

where is the DINT compatibility for the wrapper?

I had a look at the code and I can create the 'get' function, but the set_* functions are written differently. set_int returns the bytearray, and set_dword writes to the private struct.

What pattern should be used for the set_dint?

Also - with writing back to the DBRow: dynamic type detection, i dont think you can 'type' a int vs a dint - which will break the tests for writing the array's of dints (your code will think its an int).

Please let me know how you want to proceed.

Regards
Terry

spreeker, 2020-01-10:
Hi Terry,

I probably never needed to use DINT.  I think there is no problem in DB_Row / util functions since the _type variable is derived from the specification you give. So if DINT is defined in that specification, DINT functions to get and set variables will be used. Although DINT is completely missing I see.
should not be too hard. 

I would take inspiration from the INT function.

cheers,

Stephan

ETPartners, 2020-01-15:
can you please look closely to at the tests

how do you submit a DINT for testing - they are 'type' detected... and then the test harness which works for INT doesnt for DINT.

I will submit a pull request shortly - to demonstrate it.

DonatoD, 2020-04-24:
Hi, I have the same need, to read DInt value. Any suggest, workaroud?
Thanks

terryrankine, 2020-04-25:
I have it working as a direct function, but it won't work as a dictionary function.

DonatoD, 2020-04-28:
Can you explain better? Any suggestion on how to solve it?
Thanks

terryrankine, 2020-04-28:
Here is the code for a small in –  https://github.com/gijzelaerr/python-snap7/issues/124  a dint is 4bytes,  https://github.com/gijzelaerr/python-snap7/issues/20  HOWEVER – if you are using a spec/DB layout – neither of the functions will be able to be unit tested – as there is not easy way to get it into python using the current mappings. RegardsTerry  Sent from Mail for Windows 10 From: DonatoDSent: Tuesday, 28 April 2020 8:40 AMTo: gijzelaerr/python-snap7Cc: Terry Rankine; AuthorSubject: Re: [gijzelaerr/python-snap7] dint compatibility (#143) Can you explain better? Any suggestion on how to solve it?Thanks

DonatoD, 2020-04-28:
Hi erryrankine, thanks for your time, but I'm still in trouble. How can I convert it?
For example my bytearray is b'\xff\xff\xff\xbd' and it's correct value is -67
I get always wrong value.
And moreover, what about when there are characters out of the 0-F hex range?

gijzelaerr, 2020-05-18:
looks like this PR implements this: https://github.com/gijzelaerr/python-snap7/pull/159

closed 2020-05-18 by gijzelaerr