gijzelaerr/python-snap7 #145: Run/stop PLC

DonatoD, 2020-02-25:
Hi, Im able to connect to PLC without any problem with
import snap7
from snap7.util import *
import struct
from snap7.snap7types import *
from snap7.snap7exceptions import Snap7Exception
from snap7.common import check_error
plc = snap7.client.Client()
plc.connect("192.168.1.10",0,1)

but I cannot stop/run it. This are the commands I tryed
plc.plc_stop()
plc.plc_cold_start()
plc.plc_hot_start()

Any help is really appreciate. Thanks a lot

DonatoD, 2020-04-23:
Hi gijzelaerr, did you find ny help? Moreover, I found how to correctly read real and int value with get_real and get_int. How about DInt S7 type value?

gijzelaerr, 2020-05-01:
hi! sorry this issue has been open now for a couple of months without activity and is not related to the library itself, so I'm closing this issue.

closed 2020-05-01 by gijzelaerr