gijzelaerr/python-snap7 #148: Unable to run

AlexMacabu, 2020-03-31:
I'm having trouble to run any code. Probably I did something wrong in the Snap7 or Python-Snap7 installation. I'm using PyCharm. Could someone help me please? Thank you.

**My code is:**
import snap7

IP = '192.168.0.1'
RACK = 0
SLOT = 1

plc = snap7.client.Client()
plc.connect(IP,RACK,SLOT)

plc_info = plc.get_cpu_state()
print(plc_info) 

**Traceback**
Traceback (most recent call last):
  File "C:/Users/Alex/PycharmProjects/S7_Python/S7_Python.py", line 7, in <module>
    plc = snap7.client.Client()
  File "C:\RoboDK\Python37\lib\site-packages\snap7\client.py", line 37, in __init__
    self.library = load_library()
  File "C:\RoboDK\Python37\lib\site-packages\snap7\common.py", line 54, in load_library
    return Snap7Library(lib_location).cdll
  File "C:\RoboDK\Python37\lib\site-packages\snap7\common.py", line 47, in __init__
    self.cdll = cdll.LoadLibrary(self.lib_location)
  File "C:\RoboDK\Python37\lib\ctypes\__init__.py", line 434, in LoadLibrary
    return self._dlltype(name)
  File "C:\RoboDK\Python37\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 não é um aplicativo Win32 válido

Process finished with exit code 1

AlexMacabu, 2020-03-31:
I think I've installed the snap7 wrong. I've downloaded the full version of Snap7 and pasted the win32 files (snap7.dll and snap7.lib) on the Python folder. I also change the python interpreter on PyCharm. It seems that fix the issue.

Thanks anyway!

gijzelaerr, 2020-04-16:
thanks for letting us know you managed to solve the issue.

closed 2020-04-16 by gijzelaerr

geproelec, 2020-05-01:
Hello Alex ,

I have the same code than you:

import snap7

IP = "192.168.0.1"
RACK = 0
SLOT = 0

plc = snap7.client.Client()
plc.connect(IP, RACK, SLOT)

print(plc.get_cpu_state())

But, my problem is the next:

Traceback (most recent call last):
  File "C:/Users/USUARIO/Desktop/CURSO_PYTHON/PLC/PLCS71200.py", line 1, in <module>
    import snap7
ModuleNotFoundError: No module named 'snap7'

Why is te reason not found the library?? Please, could you help me?