gijzelaerr/python-snap7 #151: Not found Snap7 library, why the reason?

geproelec, 2020-05-01:
Hello Alex ,

I have the same code than you:

import snap7

IP = "192.168.0.1"
RACK = 0
SLOT = 0

plc = snap7.client.Client()
plc.connect(IP, RACK, SLOT)

print(plc.get_cpu_state())

But, my problem is the next:

Traceback (most recent call last):
File "C:/Users/USUARIO/Desktop/CURSO_PYTHON/PLC/PLCS71200.py", line 1, in
import snap7
ModuleNotFoundError: No module named 'snap7'

Why is te reason not found the library?? Please, could you help me?

gijzelaerr, 2020-05-01:
looks like you didn't install the snap7 python library.

closed 2020-05-01 by gijzelaerr

geproelec, 2020-05-01:
How I can check if the library is installed? because I have dowloand the python-snap7, and I have executed the install and it seems that all ok

But It's not possile for me working with snap7 with python

Please, could you help me?

thanks