gijzelaerr/python-snap7 #153: Problems with Snap7 and Pyhton with S7-1200

geproelec, 2020-05-01:
HI,

I'm trying to communicate with a S7-1200. The ping from PC to S7-1200 is OK

My code is the next:

import snap7

IP = '192.168.0.1'
RACK = 0
SLOT = 0

plc = snap7.client.Client()
plc.connect(IP, RACK, SLOT)

print(plc.get_cpu_state())

And the error is the next:

========================================= RESTART: C:/Users/JUAN/Desktop/programas71200.py =========================================
Traceback (most recent call last):
  File "C:/Users/JUAN/Desktop/programas71200.py", line 7, in <module>
    plc = snap7.client.Client()
  File "C:\Users\JUAN\AppData\Local\Programs\Python\Python38-32\lib\site-packages\snap7\client.py", line 38, in __init__
    self.library = load_library()
  File "C:\Users\JUAN\AppData\Local\Programs\Python\Python38-32\lib\site-packages\snap7\common.py", line 54, in load_library
    return Snap7Library(lib_location).cdll
  File "C:\Users\JUAN\AppData\Local\Programs\Python\Python38-32\lib\site-packages\snap7\common.py", line 46, in __init__
    raise Snap7Exception(msg)
snap7.snap7exceptions.Snap7Exception: can't find snap7 library. If installed, try running ldconfig

Could you help me with this issue?

Thank you

closed 2020-05-01 by gijzelaerr

gijzelaerr, 2020-05-01:
I'm sorry, but these are basic python questions and this issue tracker is not the right place to ask these questions.

geproelec, 2020-05-02:
Ok,

And please, where I can find the answers to these questions?

I have spend a lot hours and , I’m sorry but I can’t solve the problema, and I need it

Thank you

Juan

Enviado el: viernes, 1 de mayo de 2020 19:20
Para: gijzelaerr/python-snap7 <python-snap7@noreply.github.com>
CC: Dept. Técnico Geproelec <tecnico@geproelec.es>; Author <author@noreply.github.com>

spreeker, 2020-05-02:
Hi Jean,

The python code is missing the snap7 package it is using to connect to the
PLC. Installing the python snap7 code is not enough.

The exception says you are missing the snap7 library. Now you need to
install it and make sure python can find it.

So:
Install the snap7 library.  From sourceforge.

And/Or switch to Linux! :)

https://sourceforge.net/p/snap7/discussion/general/thread/b75eafb1/

Cheers,

Stephan.

geproelec, 2020-05-03:
Hi Stephan,

Problem solved

Thank you !

Juan

Enviado el: sábado, 2 de mayo de 2020 10:19

https://sourceforge.net/p/snap7/discussion/general/thread/b75eafb1/