gijzelaerr/python-snap7 #155: How to use 'read_multi_vars()' when variables is more than 20?

su600, 2020-05-08:
Hi! 
I have a problem when  variables more than 20 in `read_multi_vars(data_items)`

> snap7.snap7exceptions.Snap7Exception: b'CLI : too may items (>20) in multi read/write'

So I try to read variables by groups(each group have 20 variables) use `read_multi_vars(data_items[0:20])`, but there also has an error: 

> TypeError: byref() argument must be a ctypes instance, not 'list'

When print the type below：
```
print(type(data_items))
print(type(data_items[2]))
print(type(data_items[0:20]))
```
Got:
```
<class '__main__.S7DataItem_Array_48'>
<class 'snap7.snap7types.S7DataItem'>
<class 'list'>
```
Which means date_item[0:20] return a list, so what should I do to read more than 20 variables ?

Could you please help me to solve this problem, thank you!

spreeker, 2020-05-08:
My solution was to put all my variables in a single huge DB-block and just
every second or so read the entire thing in python using the read_db
functions.. Did 500+ variables this way, so basically
the entire PLC content.

How about doing multiple calls?

Stephan.

Senior Software Engineer.
+31 6 24 69 22 97
https://gitlab.com/spreeker/
https://github.com/spreeker/
https://www.nlx.io/
https://commondatafactory.nl/

su600, 2020-05-08:
@spreeker 
Thanks for your solution, I'll consider this solution later.

Your solution may have to do some reprogram in the PLC, my application is more focus on connecting to an old PLC without program changed, which may not suitable very much.

I think **Multiple calls** is same as what I mean **read by group**, I also done this in [PyLogix](https://github.com/dmroeder/pylogix) for Rockwell  Allen Bradley PLCs. The key problem in this issue is how to slice the `S7DataItem_Array` and retain the datetype.

Thank you again, I'll continue try to find a solution.

gijzelaerr, 2020-07-17:
Since there has been no activity on this issue I consider the question answered and will close this issue.

closed 2020-07-17 by gijzelaerr