gijzelaerr/python-snap7 #156: switch to CFFI

gijzelaerr, 2020-05-08:
CFFI is an easier method to create bindings. 

It also simplifies the binary wheels creation process, greatly simplifying the installation procedure.

https://cffi.readthedocs.io/en/latest/

nikteliy, 2021-11-26:
I tried out [pybind11](https://github.com/pybind/pybind11). I made a small [demo](https://github.com/nikteliy/python-snap7/tree/demo_pybind11) and built packages for all platforms (except Windows) using cibuildwheel. You can find the packages [here](https://github.com/nikteliy/python-snap7/suites/4472311632/artifacts/118986567). There are difficulties with the tests, but I think they can be solved. If you find pybind11 suitable, we can try it for the version 2.0.

gijzelaerr, 2021-11-28:
cool, cool, what do you think yourself? do you find it more suitable than the normal ctypes layer?

nikteliy, 2021-11-29:
Pybind11 + cibuildwheel greatly simplifies packaging workflow. But it requires experience in C++. I have no such experience, so I don't know what pitfalls to expect.

gijzelaerr, 2021-11-29:
i had good experiences with CFFI, didn't try out pybind11+ yet, so not sure how they compare. I know it is a header-only solution. 

We can also try to write a prototype for both with just 2 or 3 basic functions and see how they compare.

closed 2023-04-15 by gijzelaerr