gijzelaerr/python-snap7 #157: Drop Python2 support and add type information

gijzelaerr, 2020-05-08:


swamper123, 2020-05-26:
Do you think about Support for Python 3.8 or older? This would be important, especially for future async functions ( I may will look for in the near future).

gijzelaerr, 2020-05-26:
yes

swamper123, 2020-05-27:
So Python 3.8?

gijzelaerr, 2020-05-27:
What is your question? The software supports python 3.8 and older. Did you try?

swamper123, 2020-05-27:
On which Version of Python3 do you want to build up on. 3.6? 3.7? 3.8? Another one?
This would be neccessary to know, especially for async functions, which base on asyncio (they changed a lot between 3.4 and 3.8).

gijzelaerr, 2020-05-27:
If we drop python2 support I would say keep support for Python 3.6+ since some people still run that and it not EOL yet:

https://devguide.python.org/#status-of-python-branches

swamper123, 2020-07-06:
I never have worked with Travis, so could you drop the Python2 tests there?

gijzelaerr, 2020-07-06:
you can configure travis using the `.travis.yml` file in the root.

swamper123, 2020-07-17:
With #169, #166 and #170 Support was dropped.

Stuff you may have to do @gijzelaerr :
- A new Version Tag should be added and CHANGE file should be updated.
- I don't know if Pypy is tracking github versions, so maybe an update there as well would be good.
Then we are good to go and ready for the future. 😎

gijzelaerr, 2020-07-17:
py2 support has been dropped, but no type annotations have been added, so I'm leaving this issue open. Uploads to pypi are automated with travis. 

I'm not fully convinced of a new release yet, since of the py2 support drop and no other major changes. I'm not fully convinced yet by the maturity of the async code also, and it only implements 2 functions. Maybe we should mark it experimental for now, and see if the community comes back with feedback. @spreeker what do you think of the async Implementation?

swamper123, 2020-07-17:
The async code will be extended in the future, no worries. Next PRs about that are in my pipeline. 😄 

The async code works now, because in the older versions hasn't been any receive check and even return wrong values (zeros) if the timeframe was to short! It is still stuff to do like implementing other async methods and a way to use the callback style like in `Cli_SetAsCallback()` (eg. via `add_done_callback` of Future objects.
Waiting in idle until something is received is not an efficient way, but maybe someone (or I) can make `Cli_WaitAsCompletion()` async as well so it would be more efficient than` Cli_CheckAsCompletion()` or make such requests as Future objects which can be awaited (which has a similar effect).

But we can mark it as experimental and would realy like to hear some community feedback about that as well. :)

spreeker, 2020-07-17:
Hi I checked the code.
Nice that it is its own module. I just noticed the documentation talks about 3 modes where 3 is not supported. Raise a warning or remove option 3 from being valid.

https://github.com/gijzelaerr/python-snap7/blob/76d9832221c2caa407f9fed89ca897870088e333/snap7/client_async.py#L34
   if mode not in [None, 1, 2, 3]:

https://github.com/gijzelaerr/python-snap7/blob/76d9832221c2caa407f9fed89ca897870088e333/snap7/client_async.py#L34

swamper123, 2020-07-17:
@spreeker Thanks for that. It will be fixed in my next async PR (coming soon).

swamper123, 2020-07-22:
@spreeker @gijzelaerr Can you explain what you mean with "type annotation" exactly and what should be done?

gijzelaerr, 2020-07-22:
type hints, like:
```
def greeting(name: str) -> str:
    return 'Hello ' + name
```

swamper123, 2020-07-22:
Ah okay, I may add them in some of the next versions ... but first I have to fix some stuff I broke. 😅

gijzelaerr, 2020-11-30:
Since python2 support was already dropped made a new issue (and closed) https://github.com/gijzelaerr/python-snap7/issues/214

gijzelaerr, 2021-01-07:
implemented in https://github.com/gijzelaerr/python-snap7/pull/223

closed 2021-01-07 by gijzelaerr