gijzelaerr/python-snap7 #165: Documentation is outdated

zelitomas, 2020-06-16:
Hi,

I'm looking trough the documentation, and it seems that it has been generated from 0.7 version (commit 47b27c9f), so it lacks description of functions like `get_plc_datetime` and `set_plc_datetime`.

Also, it seems like functions with @error_wrap decorator are skipped during the generation, so functions like `connect` are missing as well.

zelitomas, 2020-06-16:
(To clarify - I'm talking about documentation on https://python-snap7.readthedocs.io)

spreeker, 2020-06-16:
So nothing stops you from adding it :).
It's free and open source build in our free time.

I tend to look at the code anyway it's not that big.

Cheers

Stephan.

zelitomas, 2020-06-17:
Well, it is not a problem of adding documentation, as it is generated from code (https://github.com/gijzelaerr/python-snap7/blob/master/doc/client.rst), but somehow triggering the build of docs for latest versions...

gijzelaerr, 2020-06-17:
the webhook was broken, fixed and updated, thanks!

closed 2020-06-17 by gijzelaerr