gijzelaerr/python-snap7 #176: OSx Tests throw errors

swamper123, 2020-07-22:
Like mentioned in #175 the OSX tests seem to explode since ... ever.
On one side I would ask if anyone uses a mac to communicate with PLCs anyway and on the other side if we want to support osx anyway, this behaviour should be fixed.

My gut feeling tells me that it has to do with the callbacks (see #174 where I mentioned that removing these two lines fix the tests again).

gijzelaerr, 2020-07-27:
I can confirm that the current master fails on OSX while running the test_server.py test suite. But when I run the 0.11 tag, there is no crash. So my guess is that it is somehow related to your changes. It would be wise to study the changes between 0.11 and master in relation to the server code and see what is going wrong. Probably something subtle changed in the C callback API, which is causing the hard segfault crash.

I'm also a bit confused why this has not been picked up by travis, since that runs the test suite on OSX also.

swamper123, 2020-07-28:
I had a look in older PRs and it seems that it was faulty before my time.
#112 (from two years ago!) has the same issue than the PRs from today. If I have a look at all Milestones from v0.11, the tests break as well in Travis.

I haven't done much with osx since years, so I may peak around for a while what it could be, but may not find a good solution.

swamper123, 2020-07-28:
I try some stuff atm and found this line in a Travis test. More infos will follow:

https://travis-ci.org/github/gijzelaerr/python-snap7/jobs/712458008#L6703
`ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/local/man'`

swamper123, 2020-07-28:
I added a line to check the modules. It seems that the snap7-server module exists in snap7-server.py.
So it can't import snap7 and so the server will never start and so every connecting try is refused.

This seems to be the issue here, but I haven't got the experience about osx to fix that.
#HelpWanted

gijzelaerr, 2020-07-28:
e8e8ddc8e44bf5f0d9a55b57a38ac6234254aff8 runs the test suite fine, while aef4dc1cd33dc0c2edba9f7185a0ceecedc478d1 fails. This means https://github.com/gijzelaerr/python-snap7/pull/169 broke the master branch. Not sure why travis didn't show us that. We need to fix this first, before we do any other changes. My guess is that it has to dome something with the types being passed around between the C call back and the python callback layer. First thing I would look at is strings and unicode type translated by ctypes lib.

gijzelaerr, 2020-07-28:
if I run it in a debugger the test_server.py test fails on the `test_clear_events` test, which crashes on line 165 in `snap7/server.py`. My guess is that the pointer is already cleared? its a bit weird.

swamper123, 2020-07-28:
Without self._read_callback and self._callback, it would be the same like before my changes. They haven't exist before, but seem to let it running ... somehow.

What I can't understand is, that these attributes exist 3 times each in the project:

    in 'init'
    in set_read_events_callback or set_read_events_callback where these are declared
    as parameter in the last line of these upper methods.

I can't see a problem here from the python side. If something is self, it should be in 'init', even if it is just None (like a good Python style). Python would even be so flexible to add these in the object, if it is declared in a method.

I can only imagine, that it can be a size problem and pointer magic. If I declare them as None, it gets an address where this variable is stored. If some c-type pointer magic may happen, which is looking for the address, it will find this space, but maybe the space size is not right. And this may happens before the interpreter (if he even notice) can handle that.

This is the only thing I can imagine.

gijzelaerr, 2020-07-28:
but they were not added by the PR right? As far as I know https://github.com/gijzelaerr/python-snap7/pull/169/ breaks master. try it for yourself, checkout https://github.com/gijzelaerr/python-snap7/commit/e8e8ddc8e44bf5f0d9a55b57a38ac6234254aff8 and run the test suite.

swamper123, 2020-07-28:
> [e8e8ddc](https://github.com/gijzelaerr/python-snap7/commit/e8e8ddc8e44bf5f0d9a55b57a38ac6234254aff8) runs the test suite fine, while [aef4dc1](https://github.com/gijzelaerr/python-snap7/commit/aef4dc1cd33dc0c2edba9f7185a0ceecedc478d1) fails. This means #169 broke the master branch. 

I had a look at [e8e8ddc] and Travis says:

[image]

(and so on with even more crashed tests, because of the server). These are the problems/crashes I mean. They still exist and existed before my first PR here, have a look at #160 and you will find the same picture and errors like this one. 

So the problem is not #169 it is something deeper and older.

> if I run it in a debugger the test_server.py test fails on the `test_clear_events` test, which crashes on line 165 in `snap7/server.py`. My guess is that the pointer is already cleared? its a bit weird.

Why is there even a `pointer`, `callback `and `library` class attribute (the stuff not inside init) which is never used?

gijzelaerr, 2020-07-28:
this is a different issue,  Actually, it is two errors. The first is the snap7 lib is not found, probably because sudo is used and the PYTHONPATH en variable is not propagated to roots environment. The second error I don't know, it seems it has problems starting the emulation server.

swamper123, 2020-07-28:
So you are talking about this?:
[image]

gijzelaerr, 2020-07-28:
i'm not sure what i shappening there, but the reason the builds on travis are not failing is because the shebang of the should probably be `#!/usr/bin/env bash -ve` or there should be a set -e (stop at error) and maybe a set -v (print command).

Now the scripts just continue after a failed command.

gijzelaerr, 2020-07-28:
I think we should fix that first, so we are sure the tests are ran properly for all supported platforms and we can move on from there.

swamper123, 2020-07-28:
So we got on our Bucketlist:

- [ ]  fixing Travis to break if a test fail
- [ ] fixing OSX tests, where something is broken since a long time, mostly based on a not starting server issue
- [ ] fixing linux test or better find the root of all evil if `self._callback` and `self._read_callback` are None and exist in __init__

gijzelaerr, 2020-07-28:
Correction, the travis build for OSX is broken, but e8e8ddc works fine on my OSX machine here.

swamper123, 2020-07-28:
I haven't got an OSX Device, but the upper error in the picture happened, because of a missing reference to the branch, which I deleted. Travis could have started again, after adding the aiounittest as dependency, and couldn't find it. Why the linux stuff hadn't got a problem with this, I don't know.

gijzelaerr, 2020-07-28:
lets see what this does https://github.com/gijzelaerr/python-snap7/pull/182

gijzelaerr, 2020-07-28:
Should be fixed by https://github.com/gijzelaerr/python-snap7/pull/182

gijzelaerr, 2020-07-29:
this has been resolved.

closed 2020-07-29 by gijzelaerr