gijzelaerr/python-snap7 #177: support for trio besides asyncio

kantrol, 2020-07-23:
**Enhancement:**
Just saw that recently asyncio support came in. It would be great if this could be extended to other async frameworks.

With [anyio](https://github.com/agronholm/anyio) there is a support library available to be independent of the async framework: 

"**AnyIO** is a asynchronous compatibility API that allows applications and libraries written against it to run unmodified on asyncio, curio and trio."

swamper123, 2020-07-27:
My opinion and philosophy about external libraries is, to use them as less as possible.
Especially if a library tries to cover three more libraries (two of them even external), it is becoming more difficult to handle something.
Asyncio is a solid built-in solution (since 3.6+) which isn't as complex as it was 4 years ago and is well supported. Nevertheless, feel free and make a fork with an anyio support solution.

Any other opinions about that?

gijzelaerr, 2020-07-27:
Yes, I agree that we should keep the scope of python-snap7 minimal.

closed 2020-07-27 by gijzelaerr

kantrol, 2020-07-29:
Ok, that's a  fair opinion.