gijzelaerr/python-snap7 #183: types.py is same name as builtin types.py

swamper123, 2020-07-30:
The [types.py](https://github.com/gijzelaerr/python-snap7/blob/master/snap7/types.py) has the same name as the [built-in types.py](https://docs.python.org/3.8/library/types.html) module.
This can lead to issues.
I would prefere to rename the module like snap7types or similar to avoid problems.

gijzelaerr, 2020-07-30:
it should not lead to issues anymore since python3 right?

swamper123, 2020-07-30:
I got this code example running:

```
import asyncio

from snap7 import client_async

async def my_read():
    # ---READ int--- Rezeptnummer
    # dbnummer, start, size
    # async def as_read_area(self, area, dbnumber, start, size, timeout=1):
    c = client_async.ClientAsync()
    c.set_as_check_mode(1)
    # adresse, rack, slot
    c.connect('192.168.150.22', 0, 1)
    g = await c.as_read_area(0x84, 1002, 0, 2)
    print(g)
    print(int.from_bytes(g, byteorder='big'))

asyncio.run(my_read())
```

Which caused:
[image]

After renaming types.py the problem was resolved.

swamper123, 2020-08-05:
Btw I used:
Python 3.8.2
Using latest Pycharm version

swamper123, 2020-08-07:
The root of the problem here is that asyncio uses the pythons `types` library. But since there is a types library in the local snap7 project, it tries to use this library. It's an import error.
Renaming the library back would be the best way. A fix is provided there.

gijzelaerr, 2020-08-09:
I can't replicate this issue. This is an issue if you put a script in the `snap7`, your current folder is in the `PYHONPATH` and you run the script as the snap7 as your working folder.

If you don't put your script inside `snap7` or don't have the `snap7` module in your sys.path or PYTHONPATH this should not happen.

gijzelaerr, 2020-08-09:
The reason it was s7types was because this used to be an issue with python2:

https://stackoverflow.com/questions/33743880/what-does-from-future-import-absolute-import-actually-do

But since we are using python3 now this is not an issue anymore. Just don't run a script from within the snap7 folder with the current folder on your python path and you should be fine.

closed 2020-08-18 by gijzelaerr

swamper123, 2020-08-18:
Sorry for the late response, did you use Python 3.8 for the testing?
Did you tried it with a testserver in another process?

gijzelaerr, 2020-08-18:
yes this is with python 3.8. i don't see why a testserver in another process would affect these results.

commit 83320c3f (2020-11-02): Add as area read write (#200)

* Rebase on Master

Add missing new line
Redo other asnyc tests with pytest
Add pytest-asyncio to requirements for tests
Replace unittests with pytests, because of version issues
Rollback to older style
Add another skip decorator about not passable test since long time
Remove empty line
Try skip test which failed in past
Try more fixes
fix possible seg fault problem
try fix test by change check_error location
Add set_as_check_mode to setup class to receive valid values
Try add async funcs in error_wrap
Fix as_ prefix at db_get
Replace unittests with aiounittests
 [... cut]