gijzelaerr/python-snap7 #185: Unable to Read

biswaKL, 2020-08-17:
Hi Team,

I have a S7-1200 with me, i cant code it. Programmer uploaded the code and left.
He gave me the register address which i am trying to read, but not happening.

these are the address:
DB1.DBW0,
DB1.DBW1,
DB1.DBW2,
DB1.DBD42

Can any one please tell me how to read these.

biswaKL, 2020-08-17:
I am getting message like:
`CLI : function refused by CPU (Unknown error)`

lautarodapin, 2020-08-17:
What code are u using to read the db?
Maybe in the plc program the blocks have check the optimized block, which will cause problems.
If you are using get_db function, try to use db_read( n_db, start, length)

swamper123, 2020-08-20:
Indeed this could have multiple issues:

- there is a typo and you try to read a space which is not available
or 
- something in your config is wrong.

[Here are the notes from snap7 about this topic](http://snap7.sourceforge.net/snap7_client.html):

> 
>  S7 1200/1500 Notes
> 
> An external equipment can access to S71200/1500 CPU using the S7 “base” protocol, only working as an HMI, i.e. only basic data transfer are allowed.
> 
> All other PG operations (control/directory/etc..) must follow the extended protocol, not (yet) covered by Snap7.
> Particularly to access a DB in S71500 some additional setting plc-side are needed.
> 1.    Only global DBs can be accessed.
> 2.    The optimized block access must be turned off.
> 3.    The access level must be “full” and the “connection mechanism” must allow GET/PUT.

biswaKL, 2020-08-20:
I reporigrammed the PLC, and i am able to read now using db_read.

Now my question is can i use: "DB1.DBD42" directly instead of giving address and size.
[image]

swamper123, 2020-08-20:
I don't know if such a feature already exist (especially if this repo shall just be a wrapper), but you may make your own methods.

You can parse yourself the string into DBnumber (1), Start (here 42), Size (DBD) (there you make yourself a dict of datasizes of word, byte etc., don't forget the endian stuff! and compare it) and this should be it. Then you throw all the stuff in e.g. `db_read()` and that should be it.

lautarodapin, 2020-08-20:
you could write yourself a function to read using "DB1.DBD42"
I suggest using regex
```python
r'DB([0-9]+)\.DB([DXBW])([0-9]+)\.?([0-9]+)?'   ##<-- starting point, you will get the DB number, Type of data, offset or byte and bit if it has
```

biswaKL, 2020-09-11:
> Did this solve your issue?

yes its resolved now. sorry i forgot to close. thanks everyone for your
help.

closed 2020-09-11 by biswaKL

M0hanraj27, 2024-09-27:
Hii., Is it Snap7 will works for acquiring real,int,dint,bool data from S7300 CP343-1 Lean module