gijzelaerr/python-snap7 #187: can't find snap7 library

emrekocak1, 2020-08-26:
Hi,

I'm using conda env (coding with pycharm), windows 10, python 3.8 and try to connect s7-1200 plc. But i get a traceback with code below.
`import time
import snap7
from snap7 import util
from db_layouts import rc_if_db_1_layout
from db_layouts import tank_rc_if_db_layout

client = snap7.client.Client()
client.connect('192.168.1.156', 0, 1)   # IP, rack, slot`

Traceback: 

`Traceback (most recent call last):
  File "C:/Users/emrekocak/PycharmProjects/veri_toplama/main.py", line 8, in <module>
    client = snap7.client.Client()
  File "C:\Users\emrekocak\Anaconda3\envs\veri_toplama\lib\site-packages\snap7\client.py", line 38, in __init__
    self.library = load_library()
  File "C:\Users\emrekocak\Anaconda3\envs\veri_toplama\lib\site-packages\snap7\common.py", line 54, in load_library
    return Snap7Library(lib_location).cdll
  File "C:\Users\emrekocak\Anaconda3\envs\veri_toplama\lib\site-packages\snap7\common.py", line 46, in __init__
    raise Snap7Exception(msg)
snap7.snap7exceptions.Snap7Exception: can't find snap7 library. If installed, try running ldconfig`

swamper123, 2020-08-26:
Have you installed snap7 (not the python-snap7) already?
Have you read the installation notice from the [docs](https://python-snap7.readthedocs.io/en/latest/installation.html#windows)?:

> Windows
> 
> Download the zip file from the sourceforce page. Unzip the zip file, and copy* releaseWindows<Win64/Win32>snap7.dll* somewhere in you system PATH, for example C:WINDOWSsystem32. Alternatively you can copy the file somewhere on your file system and adjust the system PATH.
>

emrekocak1, 2020-08-26:
First i install python-snap7 with pip. Then insalled the file from sourgeforge page and then added like in this picture. Is anything wrong here?
 
[image]

swamper123, 2020-08-26:
To be honest, I neither understand that language, nor I recognise this window. =/

What I found, what perhaps helps, ist [this](https://github.com/gijzelaerr/python-snap7/issues/56). It's windows 8.1, but I guess still a similar procedure.

gijzelaerr, 2020-08-26:
make sure everything is consistently 64 or 32 bit.

emrekocak1, 2020-08-28:
Now it shows me OSError.  My python version is 64 bit. How can i install snap7 64 bit? 

By the way at picture above is environment variables and it's windows 10, 64 bit.

swamper123, 2020-08-28:
I found the window on my windows device, but I haven't snap7 installed.
I am not sure if you have overwritten the whole PATH (this may cause problems then) already or if you edit (? I guess it's Düzenle in your window) and add the new path.
The snap7.dll and *.lib files are already in the zip folder. Snap7-full-1.4.2->release->Windows->Win64.
Alternatively someone mentioned [here ](https://stackoverflow.com/questions/33697263/python-snap7-windows-cant-find-snap7-library) to copy these files into the Python install dir (see the picture of the post).

Like gijzelaerr mentioned, check if your Python version is 64 bit as well then.

swamper123, 2020-08-31:
@emrekocak1 Did something work?

gijzelaerr, 2020-09-11:
closing since no response, assuming the problem is solved.

closed 2020-09-11 by gijzelaerr

valewyss, 2021-01-07:
Make sure you download the right file from sourceforge:
https://sourceforge.net/projects/snap7/files/1.4.2/snap7-full-1.4.2.7z/download
it should be called snap7-full-1.4.2

Snap7-VB6.zip did not work for me either, with similar error messages. 
>.... %1 ist keine zulassige Win32-Anwendung...
>“...%1 is not a valid Win32 application ...

Then copy the dll file:
release/Windows/Win64/snap7.dll to your system path

emrekocak1, 2021-01-07:
> Snap7-VB6.zip did not work for me either, with similar error messages.
> 
> > .... %1 ist keine zulassige Win32-Anwendung...
> > “...%1 is not a valid Win32 application ...
> 
> Make sure you download the right file from sourceforge:
> https://sourceforge.net/projects/snap7/files/1.4.2/snap7-full-1.4.2.7z/download
> it should be called snap7-full-1.4.2
> 
> Then copy the dll file:
> release/Windows/Win64/snap7.dll to your system path

Thanks a lot. It works!

gijzelaerr, 2021-01-07:
we hope to solve this kind of issues with the binary wheels we are trying to make.