gijzelaerr/python-snap7 #188: add snap7 library in docker image (python 3.8-buster) on raspberry pi

Yingliangzhe, 2020-08-28:
Hello guys,
I'm doing a project, that needs the snap7 to be installed in a docker image (python 3.8-buster bionic) on a raspi. 
I have tried to pull the docker image, and it is excutable on raspi (arm64). Till to is all fine.
But when I try to install the snap7 library using add-apt-repository ppa:gijzelaar/snap7, 
it occurs "no valid OpenPGP" error. 

And by excuting apt-get install libsnap7-1 libsnap7-dev, it can not locate the packages. 

Has someone already encountered this problem?

gijzelaerr, 2020-08-28:
the docker images and packages are for intel compatible architectures only, not for ARM. you need to compile it from source.

closed 2020-08-28 by gijzelaerr

Yingliangzhe, 2020-08-28:
Hello,
I didn't mean, the docker image(based on amd) from docker Hub, which you have pushed. 
Can I also use the command line :
echo "deb http://ppa.launchpad.net/gijzelaar/snap7/ubuntu bionic main" >> /etc/apt/sources.list
then the PGP key
at the end the apt-get install libsnap7-1 libsnap7-dev to install the snap7 package? 

Or must I compile the package from source in docker image?

gijzelaerr, 2020-08-28:
the same answer applies, the packages are for amd64 platforms only, not ARM. maybe the snap7 tarball already contains binaries for ARM, otherwise yes you need to compile them.