gijzelaerr/python-snap7 #190: test_read_area and test_write_area: Missing correct value check

swamper123, 2020-09-11:
In other tests are values written to the server and then checked if the read/write value was working correctly.
This is missing in `test_read_area `and `test_write_area` where just the functions are executed, but not verified of correctness. This may concerns other test methods as well.

An idea would be filling the server startup with "standard values" (values that will have always the same value (unless a write method is tested) and on the same spot in all areas) and fixing the testmethods or just local fixing the testmethods.

swamper123, 2020-12-02:
Part 1 was fixed in #200, but Part 2 is still missing (and would be still helpful).

swamper123, 2021-03-11:
At the moment, the Client is tested with the [mainloop](https://github.com/gijzelaerr/python-snap7/blob/101628dd3593d93baa3ae9085ee436c8bf438ee2/snap7/server.py#L293).

This mainloop allocates space for different kind of blocks (DB, PA, TM and CT), but not with valid datas.
In client tests, we mostly write the data to the server first and then read the values, which isn't a nice way for testing purposes.

Datatypes (hope I haven't missed or made some faults here) and typical values (mostly range, zero and some randoms between):
|Datatype|Range|usefull Values|
|---|---|---|
|BOOL|True, False|True, False|
|BYTE|0x00, 0xff|0x00, 0x1A, 0xFF|
|SINT|-128, 127|-128, 0, 100, 127|
|USINT|0, 255|0,123,255|
|INT|-32768,32767|-32768, -1234, 0  1234, 32767|
|DINT|-2147483648, 2147483647|-2147483648,  -32768, 0, 32767,  2147483647|
|UINT|0, 65535|0, 123456, 65535|
|REAL|-3,402823e+38 to -1,175494351e-38, 0.0, 1.175494351e-38 to 3.402823466e+38|-3,402823e+38,-3,402823e+12,-1,175494351e-38,-1,175494351e-12, 0.0, 1.175494351e-38, 1.175494351e-12, 3.402823466e+12, 3.402823466e+38|
|STRING[some length]|1 byte per char, depends on device and config. Std. length is 254 if not defined|length of zero, 128 and 254, filled with some easy to read string (like "the brown fox jumps over the lazy dog" or similar)|
|WORD|0x0000, 0xFFFF|0x0000, 0x1234, 0xABCD, 0xFFFF|
|DWORD|0x00000000, 0xFFFFFFFF|0x00000000, 0x123456789, 0x1234ABCDE, 0xFFFFFFFF|
|S5TIME|???|???|
|DATETIME|???|???|
|LREAL|not supported atm?|not supported atm?|

With such a fixed server, we could check all kind of values and types more easily and correctly.

We would also need some values filled with zeros (guess this would be the best way), to check write methods correctly. These values have to be set to zero again, in case the test passed.

I don't know how other areas than DB are organised, perhaps some changes are needed to check them nicely as well?!

lautarodapin, 2021-03-11:
Ok, I will try to create that initial data for the mainloop.
Currently the mainloop is generating a DB1, we could use DB0 for reading and DB1 for writing as an example.

> I don't know how other areas than DB are organised, perhaps some changes are needed to check them nicely as well?!

What do you mean with organised?

swamper123, 2021-03-12:
>  Currently the mainloop is generating a DB1, we could use DB0 for reading and DB1 for writing as an example.

This is a good idea!

> What do you mean with organised?

I neither know "what is inside"(just datatypes like in DBs as well) nor what values are allowed and what not.

lautarodapin, 2021-03-12:
> 
> > What do you mean with organised?
> 
> I neither know "what is inside"(just datatypes like in DBs as well) nor what values are allowed and what not.
DB is the only structure, that I know, that can save reals, int, booleans... Etc.
Then you have specific memory locations, like inputs, outputs, marks, counters, timers...

lautarodapin, 2021-03-16:
#259

gijzelaerr, 2021-03-19:
This has been implemented, thanks @lautarodapin !

closed 2021-03-19 by gijzelaerr