gijzelaerr/python-snap7 #191: Reading DB1 of S7-1200 shows  error "CPU : Address out of range"

Zhxhh, 2020-09-15:
I am reading  s7-1200-1214C-1AG31 whose firmware version is V3.
I close all the protection method in portol and create an unprotected DB block as snap7 documentation said.
[image]
But when I tried to use "read_area(S7AreaDB,1,0,2)" to  read it , it shows error:
`snap7.snap7exceptions.Snap7Exception: b'CPU : Address out of range'`

spreeker, 2020-09-15:
Can you read the entire db and then parse its contents? There is util code to do exactly that.

Zhxhh, 2020-09-15:
> Can you read the entire db and then parse its contents? There is util code to do exactly that.

Do you mean snap7.util? I tried the code in the example.py:
`    all_data = client.db_get(1)

    for i in range(400):                 # items in db
        row_size = 130                   # size of item
        index = i * row_size
        offset = index + row_size        # end of row in db
        util.print_row(all_data[index:offset])
`
it shows error:
` db_get => snap7.snap7exceptions.Snap7Exception: b'CPU : Function not available'
util.print_row => AttributeError: module 'snap7.util' has no attribute 'print_row'`

Yingliangzhe, 2020-09-15:
What do you want with this print_row function? 
In util.py there is a function in class DB_row named export(). That export a entire data block.  
But you need to fill the layout in the right way. In util.py test you can see, how to configurate data block layout.

Zhxhh, 2020-09-15:
> What do you want with this print_row function?
> In util.py there is a function in class DB_row named export(). That export a entire data block.
> But you need to fill the layout in the right way. In util.py test you can see, how to configurate data block layout.

The "print_row" function is defined in example.py under example folder.I just test it.
I tried `all_data = client.upload(1)` to get bytearray from plc and it still shows error:
`b'CPU : Function not available'`

Yingliangzhe, 2020-09-15:
The print_row function is not defined in example.py, it is only called in that script. 
It should be defined in util.py, but it is not. 
I tried the upload() function in class client, but it works. I just got the connection error. 
You may go into the source code of client class, and see what happens there.

swamper123, 2020-09-15:
> The print_row function is not defined in example.py, it is only called in that script.
> It should be defined in util.py, but it is not.

"print_row" doesn't exist in `util.py`, it exist in `test_util.py`.
The code in example.py is not uptodate in that case.

What I can imagine is, that your DB value may not start at DB1 0.2

Zhxhh, 2020-09-15:
> The print_row function is not defined in example.py, it is only called in that script.
> It should be defined in util.py, but it is not.
> I tried the upload() function in class client, but it works. I [...]

Can I ask what type of plc did you use? Cause when I connect to s7-300, upload function works but not work with s7-1200(v3).
Thanks in advance

swamper123, 2020-09-15:
@Zhxhh The docs say:

[image]

Zhxhh, 2020-09-15:
> @Zhxhh The docs say:
> 
> [image]

Yes , I said in my first question. I finished the setting as documentation said.

Zhxhh, 2020-09-15:
> @Zhxhh The docs say:
> 
> [image]
> 
> EDIT: So Block Up-/Download or Delete is not supported by Snap7 for 1200

emmm...I missed that. So do you know how I can read DB in s7-1200? Cause db_get() and read_area() shows error.

swamper123, 2020-09-15:
@Zhxhh 
This is my way:
0) Setup like mentioned in docs (✔️ , like you said)
1) Check if your value is realy in DB1 and starts at address 0 and is 2 Bytes big. (❓ )
2) Make your Client and client.connect("IP ADDRESS HERE"). (❓ )
3) val = c.db_read(DB_NUMBER, START_ADDRESS, SIZE) (❓ )

This is the way I do it.

Zhxhh, 2020-09-15:
> @Zhxhh
> This is my way:
> 0) Setup like mentioned in docs (✔️ , like you said)
>
> 1. Check if your value is realy in DB1 and starts at address 0 and is 2 Bytes big. (❓ )
> 2. Make your Client and [...]

I used portal to set the same DB value in s7-300 and s7-1200. 
Then my code is 
```
client=snap7.client.Client()
client.connect('192.168.1.1', 0,1)
client.read_area(S7AreaDB, 1,0,2)
````
And the code work with s7-300 but not work with s7-1200

swamper123, 2020-09-15:
Well I haven't got any other idea, except checking Step 1) again, if your value was correctly ported by TIA. 
This is what I interpret in `snap7.snap7exceptions.Snap7Exception: b'CPU : Address out of range'`.
I can't imagine another reason. 😕

Yingliangzhe, 2020-09-15:
Did you use the optimized data block? @Zhxhh 
Because since S7-1200 there is a new functionality in TIA, that optimitzes the memory. 
Maybe you can try to unclick it.

Zhxhh, 2020-09-15:
> Did you use the optimized data block? @Zhxhh
> Because since S7-1200 there is a new functionality in TIA, that optimitzes the memory.
> Maybe you can try to unclick it.

Yes, I did the three actions as doc said which contains uncliking.

spreeker, 2020-09-15:
And then make sure you applied the change into the PLC?

jay14595, 2020-09-25:
use client.db_read(dbNo, startAddress, endAddress).
Example:
 To read db1 from 0 to 96  bytes i used client.db_read(1,0,96) it works for me.

gijzelaerr, 2020-10-26:
no activity for a while in this issue so I assume the question has been answered.

closed 2020-10-26 by gijzelaerr

999sathish999, 2024-05-27:
This may occur because of trying to access the non-existing index value from the database.

If you initialize the snap7 DB size as 18, in that case The array created by the size of 18 with index value from  0 to 17.
Hence, if you tried to access the array index 18 it will throw a "CPU Out of Range Error.