gijzelaerr/python-snap7 #194: DB and DBRow can't read marks?

lautarodapin, 2020-09-19:
Hi,
The DB and DBRow class inside snap7.util can't read other thing than DB?
I'm asking because it isn't very clear on how those two class work.

swamper123, 2020-09-21:
Well, they are called DB and DB_Row, so my intuition tells me, this should only work with DBs as default.
In their methods there are some client.db_read and client.db_write commands, so this would confirm my intuition.

Why they are there and how they work exactly, I can't tell you, but I guess you shall create a class if you want to manage multiple DB values in a row, without dealing one by one. But I guess gijzelaerr or spreeker may serve a more detailed answer.

lautarodapin, 2020-09-21:
Ok.
I also though that, maybe I will fork and add some extra functionality to work with Marks, Inputs and Outputs.

closed 2020-09-21 by lautarodapin

spreeker, 2020-09-21:
Once upon a time there was a bunch of PLC's with 200+ actuators, valves, sensors and pumps connected to it. So I needed a way to get the information reliable in my python program.

So we created a DB in wich each actuators would register a few fields. Like tag, description, value, target_value, status, error. 

The DB class uses a mapping describing the data in 
Yhise tag description value etc fields. So there could be many instances of this mapping in a single huge DB.
So a DB contained many value  DB_rows. Each row representing an actuator.

So this was the intention of the code. 

You could make variations of this code to read out different kind of things but that should be some separate util class sharing some code.

Hope this clear things up.

lautarodapin, 2020-09-21:
> Once upon a time there was a bunch of PLC's with 200+ actuators, valves, sensors and pumps connected to it. So I needed a way to get the information reliable in my python program.
>
> So we created a [...]

Thansk, very clear, and it is still very useful. I just want to be able to read marks with this and treat it like DBs, because are basically the same thing xD
I'm going to fork it and add it. 
Question, should be a mixin? or addin the same class?