gijzelaerr/python-snap7 #195: Running s7server in local machine needs fix

jay14595, 2020-09-24:
In server.py i have faced the problem in line no 226     #self.set_param(snap7.snap7types.LocalPort, tcpport) after replacing it with **self.set_param(ip, tcpport)** i got my server running i connected with HMI too works fine. The default 0.0.0.0:102 not works for me only 'some_local_ip:102' works. I am newbie, pardon me if am wrong.

lautarodapin, 2020-09-26:
Change the port for the server, to 8080 or some in that range

jay14595, 2020-10-01:
Sorry above said change in line #226 is not required.
In windows the behavior is weird, but in linux it works fine. In windows when  server programs runs for the first time  it works fine on  next run several process are created with same port no 102, after deleting previous process current process works good.

Conclusion is in windows we have to manually delete process but in linux it is not required.

gijzelaerr, 2020-10-26:
closing this issue, this is more a low-level socket issue/question than a python-snap7 problem we can address.

closed 2020-10-26 by gijzelaerr